fashberg/WThermostatBeca #1: State "cooling" on heat only thermostat

erelor, 2020-04-03:
I configured the thermostat as Floor heating (BHT-002-GBLW )
On MQTT properties the thermostat send on the state field "heating" when the output is on and "cooling" when it's off.
But it should send "off" when the output is off, since this thermostat doesn't have cooling capabilities. Am I wrong?

fashberg, 2020-04-03:
Hi  @erelor ,

please check if settings (thermostate and clock) got broken after update. 
The settings structure changes from release to release, there is no settings upgrade feature yet. 

Also meanwhile there is 1.07-fas released with some more features and stability -  after Upgrade from 1.03b o 1.07-fas you have to set timezone manually  (no more cloud depandancy)

Kind Regards
Folke

Bettman66, 2020-04-04:
I have the same problem, I use this solution in WBecaDevice.h:

`
if ((isSupportingCoolingRelay()) && (state != nullptr)) {
`
`
     // cooling = digitalRead(PIN_STATE_COOLING_RELAY);
`
`
}
`

erelor, 2020-04-04:
> please check if settings (thermostate and clock) got broken after update.

Yes, I noticed broken clock configuration after upgrade from 1.03 to 1.06. 
I just upgraded firmware to 1.07 but, since the thermostat is wall mounted, I upgraded it via OTA.
Should I flash it erasing flash before to be sure it will work correctly?

fashberg, 2020-04-04:
> Should I flash it erasing flash before to be sure it will work correctly?

hi @erelor ,
OTA is fine, just check and save all settings after upgrade

erelor, 2020-04-04:
> I upgraded from 1.03 to 1.06 and now on "state" properties field the thermostat are sending "heating" when the output is on and "cooling" when it's off.

So, if OTA upgrade is ok, as I previously wrote this behaviour is a bug introduced after firmware versione 1.03.

Bettman66, 2020-04-05:
If I delete the flash memory before the update, it works. A button for deleting the flash memory at the factory setting would be helpful.

erelor, 2020-04-05:
> A button for deleting the flash memory at the factory setting would be helpful.

Or after, like the one in Tasmota firmware to reset to firmware default. Could it be implemented?

erelor, 2020-04-05:
An update. I flashed on my 3 thermostats firmware 1.07b erasing flash before. 

One is model GC and now it's working fine. Correctly send "off" and "heating" state on stat properties.
The other 2 are GB. One of these is unreachable via web after I configured wifi settings in access point mode. The other is now on the IP 192.168.2.51 but it's sending wrong information on stat/properties (each information is wrong: IP, firmware, temperature, target temperature, ...):
`{"idx":"Termo","ip":"192.168.2.11","firmware":"1.06b","temperature":19.00,"targetTemperature":16.00,"deviceOn":true,"schedulesMode":"off","ecoMode":false,"locked":false,"floorTemperature":0.00,"mode":"heat","state":"cooling"}`

Moreover, on the web schedule configuration every fileds are "0", but if I request schedules via MQTT I receive info as shown on the display interface.

erelor, 2020-04-05:
Another update. 
Also the 3rd thermostat now is reachable via web. I configured it and I receive in the properties a never seen before parameter called "floorTemperature".

This is the received message
`{"idx":"Termo2","ip":"192.168.2.19","firmware":"1.07b","temperature":23.00,"targetTemperature":10.00,"deviceOn":true,"schedulesMode":"off","ecoMode":false,"locked":false,"floorTemperature":71.00,"mode":"heat","state":"off"}`

fashberg, 2020-04-06:
Hi @erelor  and @Bettman66 !

Thanks for opening the issue.

I've fixed it in 1.08-fas https://github.com/fashberg/WThermostatBeca/releases/tag/v1.08-fas  - along with some other new features.

There was a hidden option 'supportingCoolingRelay' which was not available on the Web-GUI, but during the upgrade it got set to 'true'.  It's now visible on the Web-GUI and disabled if it's a heating-thermostat.

Kind Regards

Folke

closed 2020-04-06 by fashberg