fashberg/WThermostatBeca #6: Home Assistant Auto Config - only shows heat or off

austwhite, 2020-04-25:
This is more a feature request I guess.
With Home Assistant, when the thermostat has the hardware mod, there is the option to show Heating or Idle when the device is on and Off when the device is off.
The current implementation only shows Heat and Off and does not indicate the idle time.

Would it be possible for this to be implemented?  I do understand this may be tricky given the way the `state` attribute works in the firmware

I really appreciate the work you've done so far.

As a note, to do this in YAML manually, I had to use a script due to Klaus's firmware only using `off` and `heating` for the `state`.   Below is the YAML I had to use to get `idle` status to show in Home Assistant.

```
action_topic: 'centralheat/stat/things/thermostat/properties'
action_template: >-
  {% if value_json.state == 'heating' %}
    heating
  {% elif value_json.mode == false %}
    off
  {% else %}
    idle
  {% endif %}
```

fashberg, 2020-04-25:
Hey @austwhite ,

hmm, i see one Problem: the "mode" is a read/write property. So you can set to auto, heat and off (or cooling modes on other models)

If you set mode to "auto" for example in hass then hass expects that reading this property it returns now "auto", not "idle". But "idle" is not a state which can be set by user.

In other words:
* state cannot be controlled by user 
* mode cannot have state idle, because mode can be user-controlled

But, after some reading: https://www.home-assistant.io/integrations/climate.mqtt/

You are talking about action_topic - this is a read-only flag.
Then it makes sense. 

So i will add another read-only property called "action" with values 
* off
* idle  (only with hardware-mode)
* heating
* cooling (only BAC)
* fan (only BAC)

Does this looks suitable?

Kind Regards
Folke

austwhite, 2020-04-25:
@fashberg  Yes I am referring to the action_topic and action_template.  
I used the action_template I listed as it was the only way I could get the existing firmware to do it,

What you suggest could work very well with home assistant.
The other option, you are already using an attribute called "state" which is separate to mode,
This currently only switches between heating and off (if the hardware mod is done).
Would it be possible to make this state attribute multi-state so it can read heating/cooling/idle/off and save creating a new attribute?

austwhite, 2020-04-25:
Also, not sure how hard it would be to implement this.   These thermostats also have there "ecoMode" which could be classed as an "away" mode.

Maybe something like this could work to implement it in HA ?

```
away_mode_command_topic: "centralheat/cmnd/things/thermostat/properties"
away_mode_state_topic: "centralheat/stat/things/thermostat/properties"
away_mode_state_template: "{{ value_json['ecoMode'] }}"
```

fashberg, 2020-04-25:
Hey @austwhite 
Check out v1.10-fas 

Feedback appreciated

austwhite, 2020-04-25:
I'll check it out right now and let you know shortly :)

austwhite, 2020-04-25:
Hi @fashberg 
Just tried your new 1.10-fas and I have a little bug here.  It has created the sensor automatically, but has not created the climate entity automatically.

I might have found the bug though.
With the new action addition it is sending `action_state_tpl` and `action_stat_t`
I believe, from the Home Assistant documentation these should be as follows:
```
'act_t':               'action_topic',
'act_tpl':             'action_template',
```

Documentation: https://www.home-assistant.io/docs/mqtt/discovery/

Thanks :)

austwhite, 2020-04-25:
Just additional,  The action attribute works perfectly,  I have monitored it in a manual setup and I don't need to mess around with scripts.
Just need to work out the bug in the auto-discovery

austwhite, 2020-04-25:
@fashberg I did note one other little bug.  When the thermostat is changed from Auto to Manual and the targetTemp updates, the JSON still sends the old targetTemp until the temperature is changed either in Home Assistant or on the thermostat.
Example:
The current schedule has a temperature of 15 but the manual temperature is set to 26.  When changing to manual mode (changing auto to heat) the Thermostat turns on as it is set to 26 now, but the targetTemperature still sends 15.

austwhite, 2020-04-25:
hi @fashberg 
I don't know how to make a pull request or anything to fix the issue, but I have tested it on my own device.

I changed `action_state_tpl` and `action_stat_t` to  'act_t' and 'act_tpl' in code and compiled it myself and it now works properly with auto discovery as well.

If you have time, can you modify your code ?

austwhite, 2020-04-26:
hi @fashberg 
I've made another minor change to the Home Assistant Auto Discovery to add support for the ecoMode function in discovery.   This is Away mode in Home Assistant.
Let me know what you think of merging this to fix the existing Auto Discovery and add the ecoMode function.

closed 2020-04-26 by fashberg

austwhite, 2020-04-26:
Thanks for looking at this so quickly @fashberg .
Very much appreciated