fashberg/WThermostatBeca #7: MQTT / Network parameters being ignored

ramon-roca, 2020-04-25:
Just upgraded to the 1.10-fas 8 thermostats.
In one of them, I noticed MQTT got somehow creazy, sending a message every single second, even in the case I disable MQTT, or change the configuration to i.e. a wrong server that doesn't exists, looks like is ignoring whatever I'm setting at the network configuration, keeping the old values...
Tryied downgrading 1.9, 1.0-minimal... but same effect.
Others seems to work fine.

Can't say if that was also occuring before the upgrade.

austwhite, 2020-04-25:
Check the logging section in the thermostat settings.
Make sure it is not set to verbose.

ramon-roca, 2020-04-25:
I updated again to 1.10 and lost it, no longers gets connected to the network, neither have access to the web interface.

mosquitto_sub is still showing a message every second, don't know why, so I was confused there. 

no access at all, and looking to SSIDs, I don't see the default either, so is not in AP mode.

hard reboot doesn't help. 

the thermostat looks like working physically. showing temperature etc, but not updating the time.

there is any other way to reset the device?

austwhite, 2020-04-25:
Sounds like a bad flash.....
There is one thing, but I have never got this to work, but you can try holding the power button for 15 seconds.
The only other thing would be a manual hardwire flash if you are good with a soldering iron......

https://github.com/klausahrenberg/WThermostatBeca/blob/master/Flashing.md

ramon-roca, 2020-04-25:
15 secs holding power button don't seem to work for me.
i'm absolutely useless soldering ;(, it isn't on my best skills
let's see what I can do, but very likely i'll destroy it ;)
anyway, thanks!

fashberg, 2020-04-25:
Hey @ramon-roca ,
Have you tried this: https://github.com/fashberg/WThermostatBeca/blob/master/Configuration.md#network-recovery

Should reset network settings to default name and password 12345678

austwhite, 2020-04-26:
Curious to see how this will work also.  I have a dead device floating around but I haven't been able to get the default AP to show yet.  I might play with it again if this works.

ramon-roca, 2020-04-26:
switching off + 8 secs down key doesn't go to the blinking screen/AP mode niether  in my case

austwhite, 2020-04-26:
@ramon-roca  Sounds like you may be stuck with either trying the soldering, which is fiddly as pads will lift off with excessive heat, or you may need to replace the device.
You have multiple devices as far as I can tell and only one failed?  If this is the case, I think it is possibly a faulty device or bad flash rather than a bug with the firmware.
Since setting the logging to Warning level, I am getting no weirdness in MQTT.  Everything is smooth and only the expected messages are appearing.  
I have 6 of the floor heat models now (dry contact ones) to control dampers in each room.

fashberg, 2020-04-26:
first switch off. Display will then show nowhting. release off button... then DOWN Button for about 8 seconds.

Then MCU sends a command to WiFi Module and requests AP-Mode.
If the Wifi-Module is not running (notflashed correctly or software crashing) then nothing will happen and you have to flash with soldering and TTL Adapter.

Try also to real power-cycle! Not just power (standby) button.

@austwhite: As written this procedure only sends a serial-command to wifi module. Klaus version enables just the webserver in case it is switched of. Going back to wifi mode with default PW is an enhancement of my fork.

austwhite, 2020-04-26:
@fashberg 
Another good reason for you and Klaus to work together.
I think mine failed on the Klaus version of the firmware.

ramon-roca, 2020-04-26:
tried also with power-cycle, and still not going to AP-mode.

@austwhite , yes, I've 8 devices, 1 BAC and 7 BHT. Only failed on this BHT.

so looks i'll have to try soldering, i already destroyed one in the past while trying to hw modify to check the relay..., let's see if now i've more success ;), and if not, i'll replace this one.

fashberg, 2020-04-26:
ok. so i will close. good luck while soldering @ramon-roca

closed 2020-04-26 by fashberg