fashberg/WThermostatBeca #9: targetTemp wrong on change from scheduleMode to Manual

fashberg, 2020-04-26:
@fashberg I did note one other little bug.  When the thermostat is changed from Auto to Manual and the targetTemp updates, the JSON still sends the old targetTemp until the temperature is changed either in Home Assistant or on the thermostat.
Example:
The current schedule has a temperature of 15 but the manual temperature is set to 26.  When changing to manual mode (changing auto to heat) the Thermostat turns on as it is set to 26 now, but the targetTemperature still sends 15.

_Originally posted by @austwhite in https://github.com/fashberg/WThermostatBeca/issues/6#issuecomment-619405633_

fashberg, 2020-04-26:
You're right. This happend on manual change on the Thermostate.
Fixed in e38b25f2f01732c09e45f0fb0ceaca501be12a5e  (1.11-fas)

closed 2020-04-26 by fashberg

austwhite, 2020-04-26:
Thanks.  I'll try 1.11