fashberg/WThermostatBeca #10: Thermostat changes to Unknown multiple times in Home Assistant

austwhite, 2020-04-27:
Hi @fashberg 
Sorry to be a pain with this.  
I have noticed, with that the firmware, when using Home Assistant Auto Discovery, will keep dropping the thermostat to unknown status and messes the history graph up, as well as Home Assistant not being able to access the device when in unknown status.
This only happens when using Auto Discovery,  When using Manual MQTT HVAC setup it is fine.
I have tried downgrading, and this happens on 1.09, 1.10 and 1.11 firmware.  I do not have 1.08 to try.
As you can see from below, it will do this every 5 minutes and stay unknown for one minute before becoming available again.  Below is just a small extract from my Home Assistant log, but it shows it is regular.  Again, this only happens with Auto Discovery.  Manually added devices are rock solid.

Extract from my log file in Home Assistant:
8:00:07 PM
Central heating changed to heat
7:59:24 PM
Central heating changed to unknown
7:55:07 PM
Central heating changed to heat
7:54:24 PM
Central heating changed to unknown
7:50:07 PM
Central heating changed to heat
7:49:24 PM
Central heating changed to unknown
7:45:07 PM
Central heating changed to heat
7:44:24 PM
Central heating changed to unknown
7:40:07 PM
Central heating changed to heat
7:39:24 PM
Central heating changed to unknown

Thanks for the help

austwhite, 2020-04-27:
Hi @fashberg 
I think I have found the issue.
You are sending the Auto Config message via MQTT every 5 minutes.
This makes Home Assistant reconfigure the device via auto discovery
The Home Assistant config message should only be sent once when the device first boots as a retained message and should only be resent if the device configuration changes and Home Assistant needs to reconfigure the entity.

Can you look into this?

fashberg, 2020-04-27:
oh, i see. i will double read the mqtt docu and will fix within the next days.
I could swear my tasmotas are repeating the hass autodiscover messages.

thanks a lot for issuing the bugs @austwhite !

austwhite, 2020-04-28:
@fashberg I appreciate your time on this.  The official documentation is not clear on how often to send the discovery, but a few posts I have read suggest sending it only when you want the device configured, recommending sending as a retained message.   If the message is lost or becomes blank the device will then disappear, but it seems HA does re-add the device (or reconfigure it) whenever it receives the message.
I know you are busy.  I hope that helps whenever you have time to look at this :)

austwhite, 2020-05-03:
@fashberg 
I've monitored my Tasmota devices on the new "Elliott" version and they only send telemetry and status data as per TelePeriod.  None of the repeat the Home Assistant discovery data during use.  They only send this once on boot/Power on.

I do hope you are doing well and not too busy.
Brendan.

closed 2020-05-09 by fashberg

fashberg, 2020-05-09:
Hey Brendan!

sorry for the long delay. Had some other private projects next to this the last days.

This week my AliExpress Order arrived and now i have changed the last three old thermostates with new ones.

Thanks again for your input.  V1.12-fas should fix the issue

Kind Regards from Munich

Folke

austwhite, 2020-05-09:
I'll give it a try :). 
I appreciate your time and support

austwhite, 2020-05-09:
@fashberg Definitely fixed the issue.  Thank you

commit 55382ee5 (2020-05-09): 1.12-fas
* HASS Autodiscovery interval fix #10 (thanks again austwhite)
* installation documentation