fashberg/WThermostatBeca #12: Moes BHT GCLW

alessandrosalafia, 2020-04-27:
Hi.  I uploaded your fantastic firmware to my BHT Moes, but mine are GCLW.  I have some problems with the mqtt sectars to set the minimum and maximum temperatures of the thermostat.
[image]
[image]

austwhite, 2020-04-27:
Those are home assistant errors.  Try this as your code in yaml.
```
climate:
  - platform: mqtt
    name: Central heating
    modes:
      - "off"
      - "heat"
      - "auto"
    mode_command_topic: "centralheat/cmnd/things/thermostat/properties/mode"
    mode_state_topic: "centralheat/stat/things/thermostat/properties"
    mode_state_template: "{{ value_json.mode }}"
    temperature_command_topic: "centralheat/cmnd/things/thermostat/properties/targetTemperature"
    temperature_state_topic: "centralheat/stat/things/thermostat/properties"
    temperature_state_template: "{{ value_json.targetTemperature }}"
    current_temperature_topic: "centralheat/stat/things/thermostat/properties"
    current_temperature_template: "{{ value_json.temperature }}"
    action_topic: "centralheat/stat/things/thermostat/properties"
    action_template: "{{ value_json.action }}"
    away_mode_command_topic: "centralheat/cmnd/things/thermostat/properties/ecoMode"
    payload_on: 'true'
    payload_off: 'false'
    away_mode_state_topic: "centralheat/stat/things/thermostat/properties"
    away_mode_state_template: "{{ value_json.ecoMode }}"
    min_temp: 5
    max_temp: 35
    temp_step: 0.5
    precision: 0.5
```

austwhite, 2020-04-27:
The other option is to use auto-discovery, then the integration picks it up and you don't need manual yaml, but there is a little bug in 1.11 that does make it drop off sometimes when in auto config.  I have raised an issue for that.

austwhite, 2020-04-27:
@fashberg May need to update the readme where you have the manual yaml as HA doesn't like ~: in manual yaml.  It seems to be an auto-config only thing :)

alessandrosalafia, 2020-04-27:
Auto discovery works well but if you want to control the thermostat precisely it is better to use mqtt and compile the .yaml file now my thermostat works and I can finally untie myself from the cloud.

fashberg, 2020-04-27:
thanks @austwhite  for helping @alessandrosalafia .

Fixed in documentation 38a3b74f099c9ecaa8b277215d4379dfa256f90f

I've tried using the short version as in auto-discover json with "~" but it's not possible :(

Folke

closed 2020-04-27 by fashberg

alessandrosalafia, 2020-04-27:
It’s not the problem for me to use configuration.yaml.  I tried using "~" but didn't understand why it doesn't work.  When I correct the code I finally check the thermostat correctly.