fashberg/WThermostatBeca #13: Home Assistant graph

iMike78, 2020-05-14:
Hi @fashberg 
Do you have any idea why the HA graph is scattered? There should be a blue line with the desired temperature (as it was before), but now it shows up only on several places. This is the same on both of my thermostats since I updated from v1.00 to v1.11-fas. The other thing is, the actual heating is never shows up on the graph (it was green filling before).
[image]

fashberg, 2020-05-14:
Hey @iMike78 .

Do you use HA Autodiscover?
Looks like issue #11 - which is fixed in 1.12-fas.

austwhite, 2020-05-14:
This issue is 100% fixed in 1.12-fas

iMike78, 2020-05-14:
I just updated to v1.12-fas, and I'm testing now, thank you!

closed 2020-05-14 by iMike78