fashberg/WThermostatBeca #14: State on AC power restore

fashberg, 2020-05-14:
Derived from https://github.com/klausahrenberg/WThermostatBeca/issues/102

austwhite, 2020-05-14:
Let me know if you need anyone to test things.  :)

adrez99, 2020-05-19:
I updated to 1.12 from 1.03 and the behaviour is correct now, it wans't mentioned in the changelog but it has been fixed

fashberg, 2020-06-03:
Tested. Bug not existant. Closing issue

closed 2020-06-03 by fashberg

zawasp, 2020-12-03:
I'm using 1.19.beta1-fas and I still get this strange behaviour. On power loss from the grid, the thermostats return to ON mode, even if previously they were OFF. Also, the target temperature changes to some random value (usually higher than what I've set them. This is a problem, as I'm using them at a remote location which shouldn't be heated while I'm away. I need to constantly monitor them to switch them off if needed. The thermostats are MOES BHT-002-GBLW.