fashberg/WThermostatBeca #15: not an issue, but a question/request

iMike78, 2020-05-18:
Hi @fashberg !
I'm just wondering, will you implement the new feature what is debut in @klausahrenberg new (v1.09) fw?  I mean the "Switch back to automatic mode". 
And the question is still stands, why you guys don't work together? Both of your work are exceptional, and not too easy to make the decision which fw would be better to use. :)
Cheers

fashberg, 2020-05-19:
Hey!

Regarding Switch back to automatic mode i will implement this soon. On (corona) vacation next week.

I've made an pull request on my first changes, but @klausahrenberg meanwhile changed itself a lot (changed file stucture, etc).

I couple of days before i've asked klaus again if he's interested in merging both project. In my eyes the better solution.
see here https://github.com/klausahrenberg/WThermostatBeca/issues/99#issuecomment-619398243 

Klaus is also interested, so let's see, what will come up the next days

Kind Regards
Folke

privatesam, 2020-05-20:
Thanks for all your work on this with kalus!

fashberg, 2020-06-03:
introduced in release https://github.com/fashberg/WThermostatBeca/releases/tag/v1.14-fas

closed 2020-06-03 by fashberg