fashberg/WThermostatBeca #19: Sync Thermostat with climate entity

thundergreen, 2020-06-03:
Hey.. I sucessfully setup this device.. I even added some features not documented like fan_mode to get all function i need for my ac.

now as this thermostat is just like a dummy to control the "real" climate entity I need a setup (automation) to somehow sync those two climate entities. I came up with this:

```
  - alias: 'AC_bedroom_controller'
    trigger:
      - platform: state
        entity_id: climate.bedroom_ac_control
      - platform: state
        entity_id: climate.air_conditioner_bedroom
    action:
      - service: climate.set_hvac_mode
        entity_id: climate.air_conditioner_bedroom
        data_template:
          hvac_mode: "{{ trigger.to_state.state }}"
      - service: climate.set_hvac_mode
        entity_id: climate.bedroom_ac_control
        data_template:
          hvac_mode: "{{ trigger.to_state.state }}"
```
But the negative seideffect is that once it triggers from off to any hvac_mode like cool it turn back to off immediately and hangs in this loop without any chance to escape. I am not that familar with all this templating stuff but maybe you have an idea how to sync this climate entity with climate.ac_xxx and vise versa. Because if an automation or manually changed the mode of the "real" ac climate entity i would like it shows up in the thermostat as well.

I already found the custom_component climate group

https://github.com/daenny/climate_group

but this would lead to this error:

```
2020-05-31 22:50:51 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 284, in async_update_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 314, in _async_write_ha_state
    attr = self.capability_attributes
  File "/usr/src/homeassistant/homeassistant/components/climate/__init__.py", line 196, in capability_attributes
    data[ATTR_FAN_MODES] = self.fan_modes
  File "/usr/src/homeassistant/homeassistant/components/climate/__init__.py", line 372, in fan_modes
    raise NotImplementedError
NotImplementedError
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 284, in async_update_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 314, in _async_write_ha_state
    attr = self.capability_attributes
  File "/usr/src/homeassistant/homeassistant/components/climate/__init__.py", line 196, in capability_attributes
    data[ATTR_FAN_MODES] = self.fan_modes
  File "/usr/src/homeassistant/homeassistant/components/climate/__init__.py", line 372, in fan_modes
    raise NotImplementedError
NotImplementedError
```

issue raised here:

https://github.com/daenny/climate_group/issues/27

so this genious custom_component can't be used to "sync" or just group those two climate entities.

If you have any idea how to solve this problem, i'd highly appreciate your help. Also i'd then write a post in the forum how to set this up.

fashberg, 2020-06-04:
Hey @thundergreen ,

Interesting approach. But very HASS specific.

So you have the Beca and the thermostate both controlled by HASS?
Why two controllers for one single AC?

Your snipped says, that you trigger on state changes of climate.bedroom_ac_control.
Then you send that state to climate.air_conditioner_bedroom AND to climate.bedroom_ac_control.
But: why do you send it to the bedroom_ac_control? It already has that state.

Also ensure that setting all states is working and that all states are reported back correctly.
If you set 'cool' but the HVAC still reports 'off' the state changes back. Also this could be the main cause of the endless loop.

Can you show more details of the HASS-Config of both devices, especially beca.

climate group sounds interesting.
```
> File "/usr/src/homeassistant/homeassistant/components/climate/__init__.py", line 372, in fan_modes
    raise NotImplementedError
```
The fan_modeyou are using seems not to be implemented in the climate_group.

Kind Regards
Folke

thundergreen, 2020-06-04:
Dear Folke,

the reason why I use this thermostat to control is following.

I have setup a climate component via xiaomi IR (custom_component). This climate will send the commands to the xiaomi remote which will controll the AC via IR.

Now that I don't want to use the Dasboard to control the AC but also being able to control it manually I bought this thermostat to build ito the wall (unwired as it comunicates via mqtt with hass) 

to sync the states temp etc I need this automations. Let's say someone manually changed the temp or hvac mode on the thermostat here I want it to be reflected in hass and the other way around.

I will try with automatins then. I also got some hints but not the time to test. Once it's done I'll post my automations. 

But if you might be familiar with python you might be able to implement fan_mode into this climate group custom component? This would be really awesome and would avoid making unnecessary automations.

thundergreen, 2020-06-04:
Dear Folke. Solved like this (I know I might could have done some templating but this works perfectly)

```
https://community.home-assistant.io/t/how-to-control-multiple-thermostats-via-mqtt-but-with-one-thermostat-card/144722/5
```

fashberg, 2020-08-02:
Hey @thundergreen !
Thank's for sharing your tricky configuration.

Kind Regards
Folke

closed 2020-08-02 by fashberg

thundergreen, 2020-08-05:
Welcome