fashberg/WThermostatBeca #21: The length of the fields Hostname and Topic

sochkasov, 2020-06-13:
I saw that the length of the fields changed in the project:
- Hostname (16) 
- Topic in the MQTT section (32)
It's a breaking change in the my installation Home Assistant. I couldn't find where in the source code these restrictions are. Can you tell me where can fix it.

fashberg, 2020-06-15:
Hey @sochkasov,

yes, i've changed in 1.14-fas:
* Reading NetworkSettings support for 1.00-1.11 and 1.xx-fas Versions
* Writing NetworkSettings same way Version >=1.09 does
* Changed default Hostname to "Thermostat" (to fit hostname in 16 chars)

16 chars was also too short for me, but i wanted to be compatible to settings stored by Klaus Firmwares. If you flash FAS firmware all WiFi/Network settings are kept and also vice versa.

If you change the length from 16 to 32 all values saved after hostname (hostname is first setting, so all settings are affected) will be shifted by 16 bytes in the settings-storage. So you need to reconfigure all settings and after the first change all parameters will get filled with crap.

Length of 16 for hostname and 32 for topic prefix is in deed really short. But ressources on the ESP Chip are really limited (80 kB RAM) so keeping strings short is neccessary.

I've changed all hostnames from style  'thermostat-wohnzimmer' (living-room-thermostate) to "therm-wozi'.

If you want to change settings-structure:

Storage Size:
https://github.com/fashberg/WAdapter/blob/master/WAdapter/WNetwork.h#L1365

HTML-Form max-length
https://github.com/fashberg/WAdapter/blob/master/WAdapter/WNetwork.h#L1006

Kind Regards
Folke

sochkasov, 2020-06-15:
Thank you for the explanation. I'm going to shorten the name to support the current and future versions

closed 2020-06-15 by sochkasov

labodj, 2026-11-08:
@sochkasov I maintain a fork of this project and have released [v1.26.0-labo](https://github.com/labodj/WThermostatBeca-labo/releases/tag/v1.26.0-labo). Hostnames now support 32 bytes and MQTT base topics 64 bytes. Recognized older layouts with those longer values are migrated through a checked extension, without enlarging the original fields in place or shifting later settings. The migration tests verify the full hostname/topic and resulting discovery identity after saving and reloading.

A reset is not needed for the supported layouts. Values already truncated by an older firmware cannot be reconstructed automatically; back up long values before any downgrade to firmware without the extension.