fashberg/WThermostatBeca #24: Support for wifi (rssi) sensor

blackscreener, 2020-06-30:
Hi. Can you add wifi connection sensor to code?

fashberg, 2020-06-30:
Hey @blackscreener ,

good idea and not hard to implement.

I will implement this

Kind Regards
Folke

blackscreener, 2020-06-30:
@fashberg 
Thx, something like that:
```
int quality;
const int RSSI_MAX = -50; // define maximum strength of signal in dBm
const int RSSI_MIN = -100; // define minimum strength of signal in dBm
unsigned long lastTimeChecked = 0;

void loop() {
////////////////// TELE ////////////////////////////////////  
currentMillis = millis();
  if ((currentMillis - lastTimeChecked) > (60000 * 5)) {    ///1 min * x min // tele time
    dBmtoPercentage(WiFi.RSSI());
    lastTimeChecked = currentMillis;
    client.publish("esp/rssi", String(quality).c_str()); // wifi signal
  }
  /////////////////////////////////////////////////////////////
}

int dBmtoPercentage(int dBm)
{
  if (dBm <= RSSI_MIN)
  {
    quality = 0;
  }
  else if (dBm >= RSSI_MAX)
  {
    quality = 100;
  }
  else
  {
    quality = 2 * (dBm + 100);
  }
  return quality;
}
```

fashberg, 2020-08-02:
Hey @blackscreener,
you can check out beta release. I've just used the (negativ) RSSI Value, not percentage.
Others also do that way and it's easy to include in HASS. I have added to HASS-Autodiscovery.

[image]

Feedback appreciated

Kind Regards
Folke

blackscreener, 2020-08-02:
Thank you very much @fashberg 
Great work! RSSI sensor works great. I can convert dBm to % in Home Assistant.
```
  - platform: template
    sensors:
      wodomierz_rssi2:
        friendly_name: "Wodomierz RSSI2"
        unit_of_measurement: "%"
        value_template: >-
          {% if is_state('sensor.wodomierz_rssi', 'unavailable') %}          
            unavailable              
          {% elif states('sensor.wodomierz_rssi')|float  <= - 100 %}
            0
          {% elif states('sensor.wodomierz_rssi')|float  >= -50 %}      
            100             
          {% else %}
            {{ (states('sensor.wodomierz_rssi')|float + 100) * 2 }}
          {% endif %} 
```
[image]

closed 2020-08-02 by blackscreener