fashberg/WThermostatBeca #25: payload of True ignored

IanAdd, 2020-07-01:
HA sends true and false payloads out capitalised. Capitalised True is not recognised by your software. false and False both work, however.

Enclosing them in quotes in the HA payload is a workaround but then the state_on has to be explicitly added to the entity as the default is to match to the payload and "true" does not match the returned state of true, ie without quotes.
The example below works
```
switch:
  - platform: mqtt
    unique_id: ensuite_auto_enable
    name: "Ensuite can revert to Auto"
    state_topic: "ensuite/stat/things/thermostat/properties"
    value_template: >
      {{ value_json.switchBackToAuto }}
    command_topic: "ensuite/cmnd/things/thermostat/properties/switchBackToAuto"
    payload_on: "true"
    payload_off: false
# state normally not required to be here as it defaults to following payload values.
# but needed here as payload_on cannot be sent unquoted but state is returned unquoted.!
    state_on: true
    optimistic: false
    qos: 0
    retain: true
```

thundergreen, 2020-07-01:
@IanAdd Whre did you find this switch config. seems realy cool to use also in automations. I never got the state topic to work as my thermostat seems not broadcasting this:

IanAdd, 2020-07-01:
@thundergreen I watched the mqtt traffic and realised that the reply from the thermostat was different to what i sent to it.
Or did you mean where did i find the example i built this one from?

thundergreen, 2020-07-02:
where did i find the example i built this one from? <<<< this one :) as I couldn't find it anywhere in the docs here in this repo

fashberg, 2020-07-02:
LWT = Last Will & Testament.
Introduced in one of the last releases.
Thermostat askes MQTT broker to set this Topic to Offline if it should get unreachable.
HASS shows Thermostat offline if this happens.

Explained here: https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament/

thundergreen, 2020-07-02:
I don't need this feature.. as it does not work correctly in my setup of
mqtt server.

fashberg, 2020-07-02:
@IanAdd i will check uppercase TRUE.

But RFC says clearly:

2.1. Values
A JSON value MUST be an object, array, number, or string, or one of the following three literal names: false null true

The literal names MUST be lowercase. No other literal names are allowed.
value = false / null / true / object / array / number / string 

I recently had an issue with HASS not recognising true without quotation marks. Also not RFC compliant...

IanAdd, 2020-07-03:
> where did i find the example i built this one from? <<<< this one :) as I couldn't find it anywhere in the docs here in this repo

https://www.home-assistant.io/integrations/switch.mqtt/

fashberg, 2020-08-03:
RFC says true/false must be lowercase if it's meant as bool-value. 
But Home-Assistant ignores-this and uses True.

v1.16.beta2 now accepts True/TRUE/true

closed 2020-08-03 by fashberg