fashberg/WThermostatBeca #27: Enhancement ? Floor Temp driven

IanAdd, 2020-07-01:
Is it possible to control the BHT-002GBLW, which has the floor temp sensor, so that it operates the relay operation based upon whether the floor temp has reached the target temperature ?

Bl00d-B0b, 2020-07-24:
Hello,

yes, you can do it. check thermostat documentation: 

[gimtoji.pdf](https://github.com/fashberg/WThermostatBeca/files/4971573/gimtoji.pdf)

Code 4.

Edward

IanAdd, 2020-07-24:
Sorted, here is my solution.
```
climate:
  - platform: mqtt
    name: OZ_Ensuite_Thermostat
    availability_topic: "oz_ensuite/tele/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    action_topic: "oz_ensuite/stat/things/thermostat/properties"
#    action_template: "{{value_json.action}}"
    action_template: >
      {% if (value_json.targetTemperature|float <= value_json.floorTemperature|float) and value_json.action == 'heating' %}
        idle
      {% else %}
       {{value_json.action}}
      {%endif%}
    temperature_command_topic: "oz_ensuite/cmnd/things/thermostat/properties/targetTemperature"
    temperature_state_topic: "oz_ensuite/stat/things/thermostat/properties"
    temperature_state_template: "{{value_json.targetTemperature}}"
    # set current temp to temp of floor sensor not room temp
    current_temperature_topic: "oz_ensuite/stat/things/thermostat/properties"
    current_temperature_template: "{{value_json.floorTemperature}}"
    away_mode_command_topic: "oz_ensuite/cmnd/things/thermostat/properties/ecoMode"
    away_mode_state_topic: "oz_ensuite/stat/things/thermostat/properties"
    away_mode_state_template: "{{value_json.ecoMode}}"
    mode_command_topic: "oz_ensuite/cmnd/things/thermostat/properties/mode"
    mode_state_topic: "oz_ensuite/stat/things/thermostat/properties"
    mode_state_template: "{{value_json.mode}}"
    payload_on: "true"
    payload_off: "false"
    modes: [ "heat", "auto","idle", "off" ]
    min_temp: 5
    max_temp: 35
    temp_step: 0.5
    precision: 0.5
```

Bl00d-B0b, 2020-07-26:
Hello,

I use openhab, and I can implement the same. but still I assume it's easy to implement in next release.

Ed

fashberg, 2020-08-02:
Original Question how to enable external Sensor is answered.
Showing calculated action: idle state is handled also in #26 

So i will close

closed 2020-08-02 by fashberg