fashberg/WThermostatBeca #29: Refresh problems with Home assistance and overheating.

alessandrosalafia, 2020-07-18:
I flashed the new firmware but I encountered refresh problems with the manual commands.  When I turn on and off and off the Home Assistant thermostat does not see the change of state until I send the command from hassio.  I went back to the old version.  Also when I flashed a GALW thermostat the thermostat overheats up to 30 degrees then freezes, but I don't know if the thermostat has failed.  Before it worked with Tuya firmware.  Even now it works but until it stops due to too much temperature.

fashberg, 2020-07-20:
Hi @alessandrosalafia,

Regarding the first Problem: which version do you use and what version is the "old" version, which works?
What do you mean with manual switching on/off?   using the power button or manual  MQTT commands?

Regarding overheat:
Do you mean room temperature or overheats the device?
Or overheating protection with 30 degrees as the measured temperature of internal sensor and you have set 30 degrees in configmenu of the device to maximum temperature?

Kind regards
Folke

alessandrosalafia, 2020-07-20:
469/5000
Hello. For the first problem I saw it with v1.15fas. So I went back to the old v1.11fas which worked well for me. The problem can be seen when you switch on and off with the thermostat on / off button. If commanded from mqtt the thermostat always responds. For the second problem I'm trying to understand, I'm doing other checks, I'm not sure it's a firmware problem but it's strange because as soon as the flash was done then the thermostat started to overheat.

fashberg, 2020-08-02:
Hey @alessandrosalafia !

I've did some tests with 1.15fas and cannot find any problem with the manual on/off. It immediately shows up correctly in HASS.

Are you using HASS autodiscovery or do you use manual config? Can you share it?
Maybe there's an on/off loop or any automation.

Can you also share some Logs:
1) go the thermostat webpage and set logging to trace
2) open a cmd and start
```mosquitto_sub -h mqtt -t 'home/therm-test1/#' -v```  (replace home/therm-test1`with tour topic)

In my 1.16-fas i'm playing around with  "true" / "false" instead of true / false (quotation). RFC says: do not quote true/false/null.
Maybe here's something wrong.

What exactly means overheat? The device itself? where? how how? 

Kind Regards
Folke

alessandrosalafia, 2020-08-02:
Hi Folke.  I have done many tests and I always have some delay with 1.15.  It is not a big problem for me to stick to an old version.  I don't use self-disclosures but only manual config.  For overheating I have now come to the conclusion that my thermostat is brick.  I don't know why but loading the custom firmware it burned.

fashberg, 2020-08-02:
Hey @alessandrosalafia ,
ok, something went wrong during flash? Flashed with tuya-convert?
Have you tried to fix by flashing with soldering and TTL-Adapter?

Fixing the power on/off issue would improve project quality. Would be great if you send me the hass-configuration snippets so i can examine the problem.
 
You can also try latest  https://github.com/fashberg/WThermostatBeca/releases/tag/v1.16.beta-fas (just relased)

Kind Regards

Folke

alessandrosalafia, 2020-08-03:
Hi Folke, I will send you the snippets.  My thermostat was initially flashed with Tuya covert.  I usually use the soldering iron but I had a broken arm and it is difficult to do, so I tried but it didn't work.  Now I can use the arm and I have tried with the soldering iron.  But the thermostat made strange noises and didn't connect.  I used another one programmed with TTL adapter that worked perfectly.  Now I will try version 1.16beta.  I'm interested in seeing the RSSI signal.

fashberg, 2020-08-03:
Hey @alessandrosalafia 
https://github.com/fashberg/WThermostatBeca/releases/tag/v1.16.beta2-fas now accepts 'true' and 'True', this could solve your On/Off Problem.

RSSI manual config: Should be something like this

```yaml
sensor:
  - platform: mqtt
    name: Thermostat-RSSI
    device_class: "signal_strength"
    state_topic: "xxxxxx/stat/things/network/properties"
    value_template: "{{value_json.rssi}}"
    unit_of_measurement: "dBm"
```

closed 2020-08-10 by alessandrosalafia