fashberg/WThermostatBeca #31: Working Device - BHT-1000 GBW

matthewjporter, 2020-07-24:
Hi

Not an issue as such, justy a note to say I have updated a BHT-1000 GBW with your firmware and so far so good.

It has reported into HomeAssistant and working great

Just trying to change the time as it seems to be1 hour ahead even thought set to Europe/London

Cheers

fashberg, 2020-08-02:
Hey @matthewjporter 
Many thanks for your positive report. Have you set Temperature Precision to 0.5 or 1.0?

Regarding London Time:
If i set 
NTP Server = pool.ntp.org
Timezone = 99 (Use Daylight Saving Automatic)
DST = 0,3,0,2,60   (at last Sunday of March start Daylight Saving Timewhich is UTC + 60 minutes)
STD = 0,10,0,3,0  (switch to normal time at last Sunday of October and this is UTC + 0 minutes)

Works for me. Can you share your config and what do you expect?

If you have to above config correcly there's maybe a difference betwenn BHT-1000 and BHT-002:
There are two modes for setting the time:
Either i just can send the Local or GMT time or the Main-MCU can request Local Time or can Request UTC time.

Maybe there's a differencs with the BHT-1000 and you anly get the UTC Time.

In my trace i can see:

```
home/therm-test1/tele/log trace: Time via NTP server 'pool.ntp.org'
home/therm-test1/tele/log notice: NTP time synced: (2020-08-02 14:53:37, Weekday: 1, Epoch: 1596380017, Dst: 1585447200, Std: 1603594800, Diff: -300, Uptime: 304)
home/therm-test1/tele/log trace: Clock Recalc Timezone 1596383617
home/therm-test1/tele/log notice: NotifyNewDateTime
home/therm-test1/tele/log trace: sendActualLocalTimeToBeca 1596383617 + 0 days: 20.08.02 15:53:37 (dow: 7)
```
Can you set logging to verbose and start getting the logs via MQTT
```mosquitto_sub -h mqtt -t 'home/therm-test1/tele/log' -v```

watch for anything with clock/ntp/timezone and especially:
```
Request for GMT time sync from MCU
Request for local time sync from MCU
sendActualGMTTimeToBeca 
sendActualLocalTimeToBeca 
```

matthewjporter, 2020-08-02:
Hey, Sorry for not updating, I found the date and time settings so all 100%
and set to 99 as you mentioned

I have not tested fully as yet. I am still laying the UFH and need to wait
till for at least 14 days after I have completed the last bit of grout on
them :(

I have not set the precision but will look into it. I use HA with Appdaemon
and Schedy to control my electric heating so will be looking at doing this
for the UFH.

The plan is for the UFH to use the external floor sensor and the Electric
heating (Currently a SONOFF TH16 with AM3201) will use the BHT-1000
internal for the room temp, all controlled via schedy

richard-scott, 2020-10-09:
Ive a BHT-002 and I've had to change mine from this:

DST = 0,3,0,2,120
STD = 0,10,0,3,60 

to this:

DST = 0,3,0,2,60
STD = 0,10,0,3,0

Now I get the correct time for the UK.

closed 2020-11-01 by fashberg