fashberg/WThermostatBeca #32: ECO mode issue

Bl00d-B0b, 2020-07-24:
Hello,

I assume, when ECO mode is enable, then target Temperature should be read/only mqtt.

Now when I send ECO mode, my target Temperature set to value defined in firmware (that's correct). And I can't change it using buttons on thermostat (that's also correct), but I still have possibility to update targetTemperature value over mqtt

Ed

fashberg, 2020-08-02:
Hey @Bl00d-B0b 

while working on #28 i've already saw that bug and fixed it. Will be fixed available in 1.16-fas.

But then you will see: HASS Lovelace Climate Applet has a bug. There you can thange the temperature und it stays at your temperature. If you reload it shows again 20 degree. 
If you open the detailed view (3-dot menu, showing the temperature-curve) and setting it with up/down buttons then it immediately will jump back to 20 degree.

Kind Regards and thanks for your bug report
Folke

blackscreener, 2020-08-02:
Hey @fashberg 
I can't turn on ecoMode on newest beta in Home Assistant. In MQTT I can only see false message from ecoMode topic:
[image]

fashberg, 2020-08-03:
Hey,
at the latest v1.16.beta i've changed autodiscover from
```
"pl_on":"true",
"pl_off":"false",
```
to 
```
"pl_on":"true",
"pl_off":"false",
```

Then HASS sends not string-values true/false but instead boolean values which are True/false (mind case!)
home/therm-test1/cmnd/things/thermostat/properties/ecoMode True
home/therm-test1/cmnd/things/thermostat/properties/ecoMode false

RFC says at https://tools.ietf.org/html/rfc4627
2.1. Values
A JSON value MUST be an object, array, number, or string, or one of the following three literal names: false null true

The literal names MUST be lowercase. No other literal names are allowed.
value = false / null / true / object / array / number / string

See also: https://json-schema.org/understanding-json-schema/reference/boolean.html

BUT: In this case this is not a json-value, but just the value.
Interesting why True is uppercase and not false. Not consequent.

However:   In v1.16.beta2 true and True and TRUE are accepted!

@blackscreener:
In your screenshot if you scroll down some lines to thermostat-15135/cmnd then you will see HASS sending 'True' with Firmware-Autoconfig since 1.16.beta. Your screenshot shows only the result of the cmnd (which is the stat).

Checkout https://github.com/fashberg/WThermostatBeca/releases/tag/v1.16.beta2-fas

Kind Regards
Folke

blackscreener, 2020-08-03:
@fashberg 
In 1.16 beta 2 ecoMode works. You're right, HASS sends True/False value.
[image]

fashberg, 2020-10-11:
Fixed now in https://github.com/fashberg/WThermostatBeca/releases/tag/v1.16-fas

closed 2020-10-11 by fashberg