fashberg/WThermostatBeca #35: Invert relay logic

tvigouroux, 2020-08-16:
Hi,

Thank you @fashberg for this work. I flashed on a Bth-3000l-gclw with no problem.

The thing is my water heater has a inverted logic, where when dry contact circuit is closed, the heater is off and when is open the heater runs. My former thermostat had NC and NO connectors, but this thermostats hasn't, just NO logic. 

Is there a way to address this with sw or hw mod?

Thanks

tvigouroux, 2020-08-17:
Finally, I researched a little more my heater and discovered a way to invert the logic directly in its board.

closed 2020-08-17 by tvigouroux

fashberg, 2020-08-17:
Hi @tvigouroux ,

Good to know compatibility with BTH-3000-GxLW! Thank you for the feedback.

Relay-invertion: great, that you've figured it out.  SW mod would impossilbe, because the relay is controlled by the 2nd local (non-wifi) CPU (MCU), which is operated still the original Code.

The BHT-002-G**A**LW has Opening and Closing contact. But BTH-3000-g***a***lw seems to have only closing contact - according to product-image at https://www.amazon.de/Leepesx-BTH-3000L-GALW-Intelligentes-Heizk%C3%B6rperthermostat-Temperaturregler/dp/B07Y36XW5P

But according to manual from BTH-3000 the G***A***LW version has also Opening and Closing-Contact:
https://drive.google.com/file/d/1vNlGKKMkU1Xn83c0qR8sfK7CnMLJvtX7/view

HW hack by modifying the relay part of the thermostat could be possible. I would prefer to add a 2nd external relay with closing-contact (but eltako/finder relays are expensive and you need the space for it)

Kind Regards
Folke

tvigouroux, 2020-08-17:
Thanks for your reply. Yes, your are right about your guess on bht3000's relay. I extracted the board and found this, that as far as I researched is only a NO switch.
[image]

It was a relief to find a setting directly in the heater to invert logic because the firmware works perfect. 

Thank you