fashberg/WThermostatBeca #38: BAC-002 auto-mode

fashberg, 2020-08-20:
https://github.com/fashberg/WThermostatBeca/releases/tag/v1.16.beta3-fas
starts support for HASS Autodiscovery for BAC002.

I currently have 2 problems with BAC-002
* HASS supports only these modes:  [“auto”, “off”, “cool”, “heat”, “dry”, “fan_only”]
But BAC-002 has no full-auto-mode, you can switch between auto (scheduler) and manual and the mode can be switched between cool/heat/fan_only. How to map this to HASS. Any suggestions?
* when changing fan-speed mode switches to fan_only

Ideas?

fashberg, 2020-08-25:
Hi @thundergreen and @GGW42 

You can can test https://github.com/fashberg/WThermostatBeca/releases/tag/v1.16.beta5-fas

BAC-002 has now special HASS config, see https://github.com/fashberg/WThermostatBeca#hass-manual-configuration

**For BHT:**
mode|auto -> scheduler
mode|heating -> manual mode
mode|off-> device off
ecoMode|On/Off -> Eco On/Off

**For BAC:**
mode|auto -> DOES NOT EXISTS!
mode|heat -> mode heat 
mode|cool-> mode cool
mode|fan_only-> mode fan_only
mode|off-> device off
holdState|scheduler -> scheduler (and disables eco if enabled)
holdState|manual-> manual (and disables eco if enabled)
holdState|eco-> enables eco mode (scheduler|manual still active, but not relevant)
holdState|none-> disables eco mode (switches to mode)

thundergreen, 2020-08-25:
@fashberg I flashed the device and added new config. Will report after testing.

fashberg, 2020-08-25:
Hi @thundergreen 

BHT

thundergreen, 2020-08-25:
> Hi @thundergreen
> 
> BHT

I got a BAC :)

fashberg, 2020-08-25:
Hey @thundergreen ,

For BAC:
Most left button changes mode (cool/heat/fan).
To enable scheduler press most left and button right to it (fan) simultaneously for 2 seconds. The hand icon starts to blink. Up button set manual mode, down button enables scheduler.

For BHT:
Most left button changes between scheduler and manual

See manuals located in docs folder.

Kind regards

Folke

thundergreen, 2020-08-25:
And can i change this with homeassistant then? I mean between manual mode and scheduler? (soory still not clear ... )

fashberg, 2020-08-26:
> And can i change this with homeassistant then? I mean between manual mode and scheduler? (soory still not clear ... )

I've added some screenshots to the Docs:

https://github.com/fashberg/WThermostatBeca/blob/master/README.md#controlling-bac-in-homeassistant

Bye
Folke

thundergreen, 2020-08-26:
OK mine looks like this:

[image]

my entity attributes:

```
hvac_modes: heat, cool, fan_only, off
min_temp: 10
max_temp: 35
target_temp_step: 0.5
fan_modes: auto, low, medium, high
preset_modes: none, scheduler, manual, eco
current_temperature: 25.5
temperature: 21
fan_mode: high
hvac_action: off
preset_mode: manual
friendly_name: AC-Bedroom
supported_features: 25
```
But Preset modes not showing ...

this is my climate config:

```
  - platform: mqtt
    name: Bedroom AC Control
    availability_topic: "home/bedroom/tele/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    action_topic: "home/bedroom/stat/things/thermostat/properties"
    action_template: "{{value_json.action}}"
    temperature_command_topic: "home/bedroom/cmnd/things/thermostat/properties/targetTemperature"
    temperature_state_topic: "home/bedroom/stat/things/thermostat/properties"
    temperature_state_template: "{{value_json.targetTemperature}}"
    current_temperature_topic: "home/bedroom/stat/things/thermostat/properties"
    current_temperature_template: "{{value_json.temperature}}"
    mode_command_topic: "home/bedroom/cmnd/things/thermostat/properties/mode"
    mode_state_topic: "home/bedroom/stat/things/thermostat/properties"
    mode_state_template: "{{value_json.mode}}"
    fan_mode_command_topic: "home/bedroom/cmnd/things/thermostat/properties/fanMode"
    fan_mode_state_topic: "home/bedroom/stat/things/thermostat/properties"
    fan_mode_state_template: "{{value_json.fanMode}}"
    hold_command_topic: "home/bedroom/cmnd/things/thermostat/properties/holdState"
    hold_state_topic: "home/bedroom/stat/things/thermostat/properties"
    hold_state_template: "{{value_json.holdState}}"
    hold_modes: [ "scheduler", "manual","eco" ]
    payload_on: true
    payload_off: false
    modes: [ "heat", "cool", "fan_only", "off" ]
    min_temp: 16
    max_temp: 32
    temp_step: 1
    precision: 0.5
```

fashberg, 2020-08-26:
Hi @thundergreen ,

i have tested exactly your configsnipped and for me it's working.
Running HASS 0.114.4 (latest) on FreeBSD (FreeNAS).
Version issue?
Outdated lovelace frontend?

Nen schönen Abend

Folke

fashberg, 2020-10-11:
BAC-002 fully supported with https://github.com/fashberg/WThermostatBeca/releases/tag/v1.16-fas

GGW42, 2020-10-12:
Hi, Thanks for your continued work on this, I have just upgraded to version **v1.16-fas** from **v1.16.beta6-fas**.

I had a slight problem with the upgrade because Virgin Media had an outage at that time and the web server page couldn't be reached (Home assistant also lost MQTT messages at the same time).
I couldn't connect to the Thermostat at all, all my other devices (Sonoff switches mainly, all running Tasmota) were OK, it was just the Thermostat that was unavailable. 
Is there any reason why the Thermostat needs an internet connection to access the Local Web page or for MQTT to continue working?? Anyway once Virgin came back, I accessed the Web page and the upgrade proceeded normally.

fashberg, 2020-10-12:
Hey @GGW42 

that's strange. Except NTP and MQTT there is no external dependancy. I've set NTP server to something not existant and everything's still fine (but clock is 1970). Was your MQTT Server available and was DNS still working during internet outage?

Chrome F12 WebDebugging: 
[image]

Your smartphone may leave WiFi if there is no Internet - but because your other Sonoff devices were reachable that's not an explanation.

GGW42, 2020-10-13:
I was using my desktop PC which is wired to the router, my NAS runs the MQTT server & Home Assistant. 
The NAS is also hardwired to the router. I was using the Virgin DNS server so that wasn't accessible at the time. 
All my other MQTT devices were still accessible as were their web interfaces, just not the thermostat. Not a big problem, a few minutes later and i wouldn't have noticed the OUTAGE. I will try to investigate further when I get a chance.

fashberg, 2020-11-01:
BAC-002 issue closed.

closed 2020-11-01 by fashberg