fashberg/WThermostatBeca #40: From Tasmota to WThermostatBeca

Dandub, 2020-08-30:
Hi,
I've had a few attempts (using tuya-convert) to flash a BHT-002 and unfortunately when the attempt was successful, I didn't have the bin file into my files section so I ended up with the standard Tasmota (tasmota-wifiman.bin) that was part of the tuya convert.
Is it possible now to flash the WThermostat file over the existing tasmota?
Thanks,
Dan

fashberg, 2020-08-30:
Hey @Dandub ,

Sure, thats possible and easy, you just have to enter the command "SetOption78 1" before, see my 
documentation at https s://github.com/fashberg/WThermostatBeca/blob/master/Flashing.md#option-3-flashing-on-top-of-tasmota-or-other-free-firmware

Config gets lost so Thermostat becomes an AP and you have to reconfigure WiFi settings.

If you encounter not enough space problems use my minimal version from release 1.15, then flash 1.16-beta5.

You can also revert back to tasmota. Use the tasmota-lite.bin if you get space problems  (do never flash tasmota-minimal on devices without tasmota, i have flashed last week a tuya plug with tasmota-convert with tasmota-minimal and then you have no web gui for wifi config and you get stuck at this point. I had to solder on esp chip with 1,27mm pin distance. Yeahh. But i had success.)

Kind regards

Folke

Dandub, 2020-08-30:
Hey @fashberg 

Thanks so much. All working now. Much appreciated your help all along.

Best regards,
Dan

closed 2020-08-30 by Dandub