fashberg/WThermostatBeca #43: Maximum current temperature doesn't exceed 28 C

maximka87, 2020-09-05:
The maximum temperature of the thermostat doesn't exceed 28 C. I have done a lot of things restarted, factory defaults and reinstalles but nothing. Is anyone else having this issue.

fashberg, 2020-09-07:
Hey @maximka87 !

at HASS autodiscover WThermostat sets maxTemp to 35 degrees (https://github.com/fashberg/WThermostatBeca/blob/master/WThermostat/WBecaDevice.h#L60)
...could be the first limit.

Can you set to more than 28 at thermostate buttons directly?

Check manual https://github.com/fashberg/WThermostatBeca/blob/master/docs/BHT-002-Manual.pdf
Section:  6. System function settings

Code 6 sets maximum Temperature. Also here 35 degress is factory default.

You should check this

Kind Regards
Folke

maximka87, 2020-09-07:
Hi Sorry for the misunderstanding.
I mean the current temperature of the room. Not the heating temperature. I changed it from the code 1 of the air temperature and now its fixed. Sorry for THe trouble! Case closed!

closed 2020-09-09 by fashberg