fashberg/WThermostatBeca #44: Eco mode temperature

erelor, 2020-09-09:
There's a way to set economy temperature? It can be In the set in the options (holding the 2 left buttons when the thermo is off).
Since it return tu the default value (20 °C) every time you disconnect the thermostat from power, could be useful to set it via MQTT

fashberg, 2020-09-09:
Hey @erelor !

Do mean the thermostat-option of the temperature (set via 2 buttons holding) gets lost during powerloss?
Or is the thermostat just quitting the ecoMode?

Or is there a bug that WThermostat does not correctly reports the ecoMode after reboot/powerloss.

In General: 

WThermostat can't access the current setting of the ECO-Temperature We only get eco On/Off via tuya-protocoll.

So i've added to WThermostat: if ecomode is Active (via MQTT/webthings), then neither report the manual scheduledTemperature nor der manualTargetTemperature, but overide with (currently) Hardcoded ```ECOMODETEMP```.

You can change here: https://github.com/fashberg/WThermostatBeca/blob/master/WThermostat/WBecaDevice.h#L145

```c
#define ECOMODETEMP 20.0
#define ECOMODETEMP_COOL 26.0
```

Solution would be to add this to thermostat-config-page.

Kind Regards

Folke

erelor, 2020-09-09:
Hi, I'm talking about the "Economy temp" option (option "B" in the point 9 of the manual https://github.com/fashberg/WThermostatBeca/blob/master/docs/BHT-002-Manual-long.pdf)

The Economy Temp setting is lost at power loss and when the device is powered up its value is restored to the default value (20 °C). It will be great if you can change the ECOMODETEMP define with a variable to be set in configuration

fashberg, 2020-09-09:
Hi @erelor ,

Ok, Interesting. I can acknowledge that also my Test Moes BHT-002-GALW has the same bug.
After power Outage setting 'a' (eco On/Off) and 'b' (ecoTemp) are reverted back to default.
Not so the other settings 1-9, here all values still kept after power outage.

In the other manual https://github.com/fashberg/WThermostatBeca/blob/master/docs/BHT-002-Manual.pdf there are only Options 1-9 listed.

I can not read or set the MCU ecoTemperature, read https://github.com/fashberg/WThermostatBeca#can-you-add-xxx-as-a-feature

So it makes no sense to add an GUI Option with an configurable value of ecoTemp because this **is for Information/MQTT-Reporting only**!
If the Thermostate reverts back to factory default 20 degress WThermostat would still report via MQTT the GUI-configured value, but thermostat will operate at 20degress.

Kind Regards
Folke

closed 2020-10-11 by fashberg