fashberg/WThermostatBeca #45: WiFi connecting to specific BSSID

ilkkaolavas, 2020-09-17:
I have now 9 thermostats running on my house. My house consists of 3 buildings and is made of concrete and I need use 6 Access Points by Ubiquiti Unifi wifi to cover my house. All are using same WiFI name. Sometimes thermostat connects to wrong WiFi Access Point because I use also UAP-AC-LITE and UAP-AC-LR and LR is long range version and sometimes device connects to LR instead of LITE which is closer to thermostat.

This is one of the improvements I have been thinking: For example at esphome.io (https://esphome.io/components/wifi.html?highlight=wifi) you can specity BSSID to connect primarily. Each Access Point has BSSID, which is the MAC address of the AP's radio for that service set. I could be useful to be able to push thermostat to connect to specific Access Point by specifying BSSID. 

I don't know if this is possible to implement, but could help my setup to be more stable. I can fine tune with Wifi antenna strenght.

fashberg, 2020-10-13:
Hey @ilkkaolavas 

it would be easy to handle by using the bssid 
https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/station-class.rst#begin

used here: https://github.com/fashberg/WAdapter/blob/master/WAdapter/WNetwork.h#L305

But it needs gui and configuration enhancement and in case of hardware changing an AP it cannot connect to a new or other AP.

Some Vendors (AVM FritzBox with "Mash", i have two of them at home or at Enterprise Level HPE/Aruba with the Feature Client Match) are kicking the Clients if they do not Roam itself.

What about such simple solution: 
https://community.ui.com/questions/Bind-client-to-AP/a773629e-1ac9-456a-8554-7152d0406c3a

Kind Regards
Folke

closed 2022-11-08 by github-actions[bot]

labodj, 2026-10-27:
@ilkkaolavas  I’ve implemented this in my [maintenance fork](https://github.com/labodj/WThermostatBeca-labo).

Network Configuration now has an optional AP BSSID field. Leaving it empty keeps automatic AP selection; setting it locks both the initial connection and reconnects to that AP.

To address the AP-replacement concern raised above, a locked device opens its recovery AP after 120 seconds continuously offline, even when ordinary AP fallback is disabled. The five-minute configuration window lets you change or clear the BSSID; opening Network Configuration renews it.

The reconnect and recovery paths have native regression coverage. A real AP-failure/replacement test is still pending, so feedback on that scenario would be particularly useful.