fashberg/WThermostatBeca #46: Integration in iobroker

tim4849, 2020-10-06:
Hello, I have 2 BHT-002 GALWs with firmware 1.15-fas running. I also flashed a third one, with version v1.16-beta6-fas which after setting parameters does no longer reports via WLAN. Seems to be damaged, if someone is interested, I would like to make it available for test purposes.
Now my problem: I have integrated both into my iobroker using an MQTT adapter and can read everything. Unfortunately, I cannot write access, regardless of whether it is for individual values or in JSON format. No value can be set.
What am I doing wrong?

fashberg, 2020-10-13:
Hey @tim4849 ,

Can you try to update to 1.17-fas. You have a Live WebLog and can do better diagnose what's going on.
Send a snipped of the logs if you need further help.

For the broken one: Have you tried to switch off and long-press the most right button for 8 secs (https://github.com/fashberg/WThermostatBeca/blob/master/Configuration.md#network-recovery)

Using a serial adapter it can be reflashed easily. From where are you from? you can contact me at folke AT ashberg DOT de and we can see what to do.

Kind Regards
Folke

tim4849, 2020-10-16:
Hello Folke,
Thanks for the answer. I have updated to 1.17-fas, the behavior remains the same. I once attached the live WEBLog.
Thanks in advance!
[wt_63.log](https://github.com/fashberg/WThermostatBeca/files/5393371/wt_63.log)

Master-Rudi, 2020-10-16:
I have the same issue

tim4849, 2020-10-17:
I afraid you can not see any MQTT set operations in the attached log.

Master-Rudi, 2020-10-17:
I tried a lot. It´s possible to change the values with MQTT.fx in my ioBroker:

I publish to:
`_Thermostate/Kinderzimmer/stat/things/thermostat/properties/_`

That:
`{"idx":"Kinderzimmer","ip":"192.168.2.220","firmware":"1.17-fas","switchBackToAuto":true,"temperature":21.00,"targetTemperature":18.00,"deviceOn":true,"schedulesMode":"off","holdState":"manual","ecoMode":false,"locked":false,"mode":"heat","action":"heating","state":"heating","mcuId":"IAYz2WK1th0cMLmL1.0.0"}`

In my ioBroker the payload changes to my published targetTemperatur 18.00, but the thermostate stays on 23°C. 

In the log i found that:
`trace: Received MQTT callback: 'devices/thermostat'->'{"url":"http://192.168.2.220/things/thermostat","ip":"192.168.2.220","topic":"Thermostate/Kinderzimmer/stat/things/thermostat"}'
notice: Ignoring, starts not with our topic 'Thermostate/Kinderzimmer'`

Maybe there is the mistake?

tim4849, 2020-10-18:
@ Master-Rudi
How do you proceed the command in your iobroker. Via editing in "objects" (see attached screenshot) or by script?

no matter what I try, I don't get any "received ..." entries in my log

tim4849, 2020-10-18:
@Master-Rudi 
now it works (by JavaScript: 
sendTo('mqtt.1', 'sendMessage2Client', {topic: "home/studio/cmnd/things/thermostat/properties/targetTemperature", message: "23"});

Thanks for your advice!!

tim4849, 2020-10-18:
@fashberg 
network recovery does not work, I tried all, but I´m afraid the WIFI erlectronic is damaged.
If you want, I can send the thermostat to you, if you want ist for testing purposes.

BR

tim4849, 2020-10-18:
@fashberg 
Hi, I added the data points for cmnd manually and now the setting of values also works in the object tab. 

Now I've played around a little and noticed that every data point that I have set is newly created in the ioBroker under _stat/thermostat/...._ Actually, the data points under _stat/thermostat/properties/..._ should be overwritten. 

Even if I change values in the WEBConsole, e.g. adjust the scheduler, I get a new datapoint _stat.things.thermostat.schedules_

See the screenshot for this. I think this is a bug.
[image]

tim4849, 2020-10-19:
Set _false_ to _.../cmnd/things/thermostat/properties/locked_ works fine, thermostat is locked. But in WEBConole _/info?_ an also _.../stat/things/thermostat/properties/locked_ shows still _true_

This seems to be a bug.

tim4849, 2020-10-27:
Hello everyone, I installed 1.18fas, but in my opinion following bugs are still there.

Here again a summary description of the SingleProperty issues:

Example 1: 
	set _cmnd.things.thermostat.properties.deviceOn_ to false
		device switches to _off_ state
	Bug: if first set: a new datapoint _stat.things.thermostat.deviceOn_ is created
	Correct: datapoint _stat/things/thermostat/properties/deviceOn_ should be set! 
Same behaviour with following datapoints: _ecoMode, locked, mode, schedules, schedulesMode, switchBackToAuto_

Example 2: 
	set _cmnd.things.thermostat.properties.locked_ to true
		device switches to _locked_ state
	Bug: if first set: a new datapoint _stat.things.thermostat.locked_ is created, but state is always _true_
	Correct: datapoint _stat/things/thermostat/properties/locked_ should be set! 
Same behaviour with following datapoints: _switchBackToAuto_

All settings that are made directly on the device end up on the correct datapoints

Hope that clarifies this.
Best regards

PS:	
Question to Property ecoMode: Where can I define temoeraturevalues for eco?
Question to Properties mode and state: What is the difference?

Master-Rudi, 2020-10-28:
Hi there,

i have the same issues like you:

Example 1: 
i think there is a problem in the compatibility from HASS to ioBroker which results a wring datapoint.

Example 2:
The issue #53 solves this problme. I compiled a new .bin yesterday with the corrected value and now the state turns to "false". If someone is interested, i can share my .bin.

So for me it would be great if someone could find the problem mentioned in example 1

tim4849, 2020-10-29:
@Master-Rudi 
Thanks for your acknowlegement, may be we both are the only one using ioBroker? Would b egreat to get your .bin
BR

Master-Rudi, 2020-10-31:
@tim4849 

Yes i think so. too. Maybe we are the only using ioBroker. I tried to find a solution for the datapoint which are in the wrong folder.

Attached my .bin:
[wthermostat-1.18a-mr.zip](https://github.com/fashberg/WThermostatBeca/files/5468775/wthermostat-1.18a-mr.zip)
It´s the 1.18 from fas with only one bugfix from #53. Have fun and would be great if u can give me a short feedback, it´s working or not.

tim4849, 2020-10-31:
@Master-Rudi 
Yes, locked state now changes correctly - thanks!
Can now also compile, but have not yet found the place where the wrong objects are created. Would be great if you could keep me up to date!

@fashberg: we need your help and would be grateful for a solution!

tim4849, 2020-11-02:
I now have a workarround that solves the problems with the _single properties_:

- a script parses the JSON string in _home/diele/stat/things/thermostat/properties_
- objects for the _single properties_ must be created manually under _home/diele/stat/cmnd/.._ or can be imported into the 
iobroker (attention, my instance is _mqtt.1_)

For me it works so far, I don't need the _single properties_ anymore.

Have fun with it, here are the JS scripts and the import for the objects.
[mqtt.1.home.studio.cmnd.json.txt](https://github.com/fashberg/WThermostatBeca/files/5477168/mqtt.1.home.studio.cmnd.json.txt)
[wt-studio-parse-json.js.txt](https://github.com/fashberg/WThermostatBeca/files/5477169/wt-studio-parse-json.js.txt)

Master-Rudi, 2021-10-21:
The easiest way to do this is with the mqtt adapter. But here you have to make the cmnd structure manually to control it:

[image]

when it’s working with mqtt then you are able to connect and control it with yahka!

closed 2022-11-08 by github-actions[bot]

mczeus, 2024-03-22:
@tim4849 
is it possible to get the script? I just can't get it to send data to the thermostat :(

tim4849, 2024-03-24:
Hi,

sorry, it’s years ago. Can you please describe in detail what script you mean?

BR Tim

mczeus, 2024-03-24:
I have now been able to insert the data points with the MQTTX app.
I thought you had a script to add the data points.

Thank you anyway