fashberg/WThermostatBeca #47: Thermostat precision

richard-scott, 2020-10-09:
I've got a BHT-002 and I have the Thermostat precision set to 0.5. On the LCD screen I can change the target temperature by 0.5 at a time which is great.

My question is that I have to set the target temperature to 1.0 above the current temperature before it switches the heating on. Is there a way to change this down to 0.5?

I have noticed that if you switch the unit off, and back on again with the target temp being 0.5 above current temp then the heating is switched on.

richard-scott, 2020-10-09:
I've just noticed that if I switch the unit off via MQTT and back on that the unit doesn't activate my heating, but if I go to the display, press the round button off, and back on again, then it does switch the heating on.

Hedgehog57, 2020-10-09:
BHT-6000 C. Similar issue. Current setting is 22 degree. Current state is 21.5. mode - auto. Thermostat does not give command to start boiler. If I turn it (thermostat) off and on by button it will start heating.

fashberg, 2020-10-11:
Hey @richard-scott  and @Hedgehog57 

that's not a bug, that's the specified behaviour!

It's the configured deadzone/deadband. See https://en.wikipedia.org/wiki/Deadband

It can be configured between 1 to 5 (step 1.0)

The actualTemperature must be lower or equal than targetTemperature - DeadZone Temperature.

If you want the heating to start to heat at 21.5 set target to 22.5. But then it will stop at 22.5.
 
I cannot change the behavior - see:  https://github.com/fashberg/WThermostatBeca#can-you-add-xxx-as-a-feature 

> Can you add xxx as a feature?
> 
> I have been often asked if I can implement a specific feature. Often the answer is: Because all internal routines (reading temperature, controlling the relay, reading the buttons, controlling the display) are controlled by MCU, which still runs proprietary closed source software, the behavior cannot be changed!

If you really want to start to heat at 21.5 you can put logic into home assistant by using automations. You can adjust targetTemperature depending on actualTemperature (set targetTemp to 22.5 if actualTemp<=21.5, set targetTemp to 22.0 is actualTemp>21.5 or more complicated!

Kind Ragards

Folke

Hedgehog57, 2020-10-11:
Hi @fashberg,
Thanks for awesome work.
If that is a standart behavior - that's ok.
P.S. Good idea with HA.  ;)

richard-scott, 2020-10-12:
ok, thanks for the update, and nice work on the project :-)

closed 2020-10-12 by richard-scott