fashberg/WThermostatBeca #48: Can the Tuya thermostat be used without connecting a heating/cooling system?

jan--m, 2020-10-11:
Is it possible to use the BAC-1000 for two things only (without actually connecting an HVAC device):
- letting the end-user view the room temperature
- letting the end-user set a target temperature (that will be used by a virtual thermostat in home assistant)

I understand that the MCU (which we have no control over) decides what to show on the display. Will an error be displayed if it has nothing to switch on and off? And I assume that modifying the target temp from HASS is not a problem?

fashberg, 2020-10-11:
Hey @jan--m 

Yes, sure, this should be possible!
Just connect Thermostat to 230 Volt Power - the relay will not be sad if it will not switch any current  

Kind Regards
Folke

jan--m, 2020-10-12:
Thanks for the quick response @fashberg !

On a related note, do you have any idea if your firmware will work with this simple tuya thermostat?
https://www.aliexpress.com/item/1005001308144081.html
I need about a dozen of these so that $25/piece would be reasonable.

I see that somebody asked about the same thermostat here with no response:
https://community.home-assistant.io/t/wifi-thermostat-beok-floureon-beca-energy-component/48519/451

And there's another question about similar one (Beok TGP51WIFIG-EP) a few posts up but also no response...maybe I'll bump the thread.

closed 2020-10-12 by jan--m