fashberg/WThermostatBeca #49: Cannot flash BHT002

rumoldus, 2020-10-11:
Hi 
I am trying to implement this great project in my new BHT-002 thermostat. PCB matches exactly with photo (including part number). Setup is with soldered wires. I have added external 3.3V, all grounds (USB, Power supply, ESP) are common.
I am working from a Windows laptop. 
When I connect Putty (Serial, COM7, baud rate 78000), and reboot the unit with GP00 pulled down, I get the line:
    `ets Jan 8 2013, rst cause:2, boot mode (1,6)`
Looks good to me. Tells me communication is good (at least the transmit part) and dat de ESP8266 is activated in write mode.
When I try to read firmware (or read chip_id, or flash_id) I always get a time out on the esptool command.
(Using Esptool.py 2.8:
`     A fatal error occurred: Failed to connect to Espressif device: Time out waiting for packet header.`
I played with many baud rates, --no-stub option, I now even added an external pull up resistor to GP02, tried different esptool commands, but nothing seems to help. One of my concerns is that no data arrives from USB to ESP, but I do not know how to check that.

When I connect Putty and perform a normal reset (no pulling down of GP00) I get:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x40100000, len 2592, room 16
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : QIO
SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000

OS SDK ver: 2.0.0(e8c5810) compiled @ Jan 25 2019 14:26:04
phy ver: 1055_9, pp ver: 10.7

rf cal sector: 121
idle_task_hdl : 40107a90,prio:0, stack:384
tim_task_hdl : 40107bd8, prio:2,stack:512

```
This should be the original software as I have not managed to write the new firmware.
Any tips, suggestions are welcome.

fashberg, 2020-10-11:
Hey @rumoldus 

it seems the device is not in flashing mode (because you see it booting).

> https://github.com/fashberg/WThermostatBeca/blob/master/Flashing.md#22-connection-to-device-for-flashing 

Blue right: ESP-GPIO0, must be connected with GND during power up

Please double check this. If you erase first then you have to put GPIO-0 to GND again because otherwise ESP will do a normal boot after erase.

If you see something like this:

[image]

That's the binary serial communication between ESP and MCU - changing baudrate will not result in readable strings ;)

Kind Regards
Folke

rumoldus, 2020-10-12:
Hi Folke,
many thanks for your very quick answer. Appreciated.
For your information, I have all wires connected exactly as per your photo. 
I tend to disagree. When I pull down GPIO-0, I get a different boot message (the first single line, with boot mode(1,6), you may have missed it).
When I do not pull down GPIO-0, I get the long booting sequence with boot mode(3,x)
Both boot modes are in line with:
`https://github.com/esp8266/esp8266-wiki/wiki/Boot-Process#esp-boot-modes`
I'll resolder the wires anyway, but as the unit is reacting to the pull down, I do not expect a lot from it. 
I am still concerned that the Rx of the ESP8266 is not functioning.
Best regards
Marcel

fashberg, 2020-10-12:
Hey @rumoldus ,

oh, sorry. I've missed that part of your mail.

Another possibility is too less power. The USB to Serial converter must have enough power to power the ESP Chip including WiFi.

If I connect the ESP without programming-mode it is fully functional - not the LC-Display and the MCU (powered with 5 Volt), but i can connect using http to WTHermostat.

The same for example for tasmota-flashing shelly devices. I was not able to flash until i connected separate 3,3 Volt. Then i've bought better Serial Adapters (see my links to at flasing.md)

Kind Regards
Folke

rumoldus, 2020-10-12:
Hi Folke,
you missed another part of my mail. I already have 3.3 V from an external source. With this USB / Power setup I have been able to flash other 8266's. 
I have a question for you. The GPIO-02 should normally be pulled up. You don't do that in your setup, and yet it works. I did add an external resistor, when I did not succeed without one,  (pin next to the 2 black wire ones) but that didn't help me. 
I'll experiment some more and will let you know when I find my fault.
Best regards
Marcel

fashberg, 2020-10-12:
Hey @rumoldus ,

have you flashed other ESPs with that adapter successfully?

If you connect your serial adapter's TX to its RX you should see an echo in Putty - to check the adapters functionality.

Some more to read:

* https://tasmota.github.io/docs/devices/TYWE3S/
* https://tasmota.github.io/docs/TuyaMCU-Devices/
* https://www.fanjoe.be/wp-content/uploads/2018/11/Users-Manual-3525098.pdf

Spec says: GPIO2 Output UART0_TXD (used to print module's internal information)
But nothing about GPIO2 regarding programming mode

Kind Regards
Folke

rumoldus, 2020-10-13:
Hi Folke, thanks again for the quick answer.
I already mentioned that this setup has worked on other ESP's before. The Rx/Tx test I did before (did NOT mention that earlier) and the USB converter is OK. 
I have now desoldered the Wifi unit and am now able to flash it! The references you gave were useful, they mention that the MCU on the PCB may keep the Rx line busy. Sounds reasonable, except that it didn't bother you. 
I hope I can get the Wifi unit back on as I possibly damaged the 'motherboard'.
Thanks for thinking with me.
Best regards

closed 2020-10-13 by rumoldus