fashberg/WThermostatBeca #50: Automatic mode doesn't report temperature

blackscreener, 2020-10-12:
Temperature in auto mode is report from manual mode to Home Assistant. Of course RelayStateCalculation isn't work properly for automatic mode.  Also thermostat turns on heating even for 0,5 C difference when powering on by manual button.

fashberg, 2020-10-12:
Hey @blackscreener 

should be fixed with 1.17-fas

Thanks for your feedback!

Folke

blackscreener, 2020-10-13:
Thank you. Great work!

blackscreener, 2020-10-13:
@fashberg 
In 1.17 auto mode switch behavior is better but arrows responsible for adjust thermostat temperature in Home Assistant changes heat mode to manual from auto only in Home Assistant.

fashberg, 2020-10-13:
Hey @blackscreener ,

you mean that changing the temperature on the Device itself changes _not_ from scheduler to manual mode?

This behaviour can't be changed, because it's controlled by MCU! See my note on this: https://github.com/fashberg/WThermostatBeca#can-you-add-xxx-as-a-feature

Kind Regards
Folke

blackscreener, 2020-10-14:
> Hey @blackscreener ,
>
> you mean that changing the temperature on the Device itself changes _not_ from scheduler to manual mode?
>
> This behaviour can't be changed, because it's controlled by MCU! See [...]

No. When thermostat is in auto mode (scheduler) and I change temperature in Home Assistant than the mode in HA changes to manual but mode in thermostats stays on scheduler.

closed 2020-10-25 by fashberg

labodj, 2026-11-08:
@blackscreener I maintain a fork of this project and have released [v1.26.0-labo](https://github.com/labodj/WThermostatBeca-labo/releases/tag/v1.26.0-labo). For the later comments here about HA changing to manual while the device stays in auto: it fixes the missing manual-mode command when changing the target from Home Assistant/MQTT while scheduling is active. The requested temperature is saved first, then the firmware sends the manual-mode frame followed by the new setpoint frame. This keeps the ESP's reported mode and the command sent to the thermostat MCU consistent.