fashberg/WThermostatBeca #51: Home Assistant autodiscovery: auto to change some parameters like max-temp? Why is not loaded automatically?

dmerlani, 2020-10-14:
Hi, I am using WThermostatBeca and enabled Autodiscovery. Everything works fine. I have changed the max_temp on the thermostat to 23°C but the WThermostatBeca still puts 35°C. I also checked the payload and in facts sets 35°C. So it seems WThermostatBeca is not loading max temperature from the thermostat.
I could also ovveride it in home assistant but, as the code is not present in configuration.yaml I don't know how to do...

_name: thermostat-22524
unique_id: thermostat-22524_climate
device:
  identifiers:
    - 'CC:50:E3:22:5E:7C'
  name: thermostat-22524
  model: WThermostat
  sw_version: 1.17-fas
  manufacturer: WThermostatBeca
availability_topic: home/pianoterra/tele/LWT
payload_available: Online
payload_not_available: Offline
action_topic: home/pianoterra/stat/things/thermostat/properties
action_template: '{{value_json.action}}'
mode_command_topic: home/pianoterra/cmnd/things/thermostat/properties/mode
mode_state_topic: home/pianoterra/stat/things/thermostat/properties
mode_state_template: '{{value_json.mode}}'
away_mode_command_topic: home/pianoterra/cmnd/things/thermostat/properties/ecoMode
away_mode_state_topic: home/pianoterra/stat/things/thermostat/properties
away_mode_state_template: '{{value_json.ecoMode}}'
temperature_command_topic: home/pianoterra/cmnd/things/thermostat/properties/targetTemperature
temperature_state_topic: home/pianoterra/stat/things/thermostat/properties
temperature_state_template: '{{value_json.targetTemperature}}'
current_temperature_topic: home/pianoterra/stat/things/thermostat/properties
current_temperature_template: '{{value_json.temperature}}'
payload_on: true
payload_off: false
min_temp: '10'
max_temp: '35'
temp_step: '0.5'
modes:
  - heat
  - auto
  - 'off'
platform: mqtt_

sf

fashberg, 2020-10-14:
Hey @dmerlani ,

Thank you for the feedback.

The ESP chip, where WThermostat is runnning on, can't access all data of the MCU, which is still closed source.

There is no possibility to query such settings! So I just send 35 degrees  as maxTemp at HASS autodiscover to not really be the bottleneck. 

Kind regards

Folke

dmerlani, 2020-10-15:
Hi, understood and thanks for the answer. But is there no way to ovveride the 35°C as maxtemp inside homeassistant? As it's in autodiscovery mode, there is no yaml code so I don't know how to do it.
BR

closed 2020-10-16 by dmerlani

sinegovsky-ivan, 2020-11-14:
@fashberg hi! 
Thanks for your work!

Can you suggest me how to set max temp without hex-hacking of the firmware file?

Maybe has a way to disable auto-discovery and set parameters (like max-temp) manually?