fashberg/WThermostatBeca #52: Wifi SSID is case sensitive ?

rumoldus, 2020-10-14:
Hi,
a low priority issue, but I discovered that Wifi SSID must match the actual case (I happen to have an uppercase SSID and entered lower case). I got it working, but it cost me some time.
Best regards
Marcel

fashberg, 2020-10-14:
Hey @rumoldus ,

Sure, the SSID must be entered exactly. Also the WPA preshard key gets derived from SSID and secret so after changing the SSID the PSK is another one.

Do you see any mistake in the docs or software, ot just for information?

Kind regards

Folke

rumoldus, 2020-10-15:
Hi Folke, 
As I was not used to entering SSID manually, (usually you select from a list) I was not aware of this requirement. Suggest to add a line in the configuration.md.

Marcel Payens

commit cbec35e2 (2020-11-01): Doc Updates

* also fixes #52
commit 6b9395a3 (2020-11-01): v1.19.beta1-fas

* Fix/Enhancement #54: If enabled, all single Properties are sent out every notification interval, not only if changed
* Fix #52: Documentation on SSID and renamed Wifi to WiFi
* Fix #53 and #64: Boolean Values wrong
closed 2020-11-01 by fashberg