fashberg/WThermostatBeca #55: Mode after switch off and on

dmerlani, 2020-10-18:
Hi, I saw this behavior. I put the thermostat in auto mode. Then switch off (unplug the thermostat from power supply) and switch on again. It goes to manual mode, so it doesn’t remember the last state before switching off.
Could you kindly check? Thanks

Master-Rudi, 2020-10-21:
There is a bug in the configuration:

[image]

After unchecking the checkbox and save and reboot, the checkbox is active again.

dmerlani, 2020-10-21:
Hi, the checkbox is actives but this is related to “auto mode from manual at next schedule period change” and not “keep the current mode after switch off of the thermostat”. It’s different. Thanks

dmerlani, 2020-10-26:
Anyone else experiencing this issue? If i leave the thermostats in auto mode and the power supply goes down for some reasons, then why should they restart in manual mode and 20C? Then the heater will start without any reason and this is not correct.
thanx

blackscreener, 2020-10-26:
> 
> 
> Anyone else experiencing this issue? If i leave the thermostats in auto mode and the power supply goes down for some reasons, then why should they restart in manual mode and 20C? Then the heater will start without any reason and this is not correct.
> thanx

Yeah, it is a big problem. In 1.16.beta6 it was fine.

chipmoca, 2020-10-27:
> Anyone else experiencing this issue?

Yes, I have the same problem. Only after a power outage, the thermostat remains off, you have to turn it on manually.

fashberg, 2020-11-01:
Hey @dmerlani,  @Master-Rudi , @chipmoca and @blackscreener 

Thank you for your findings.

Regarding the not saved “auto mode from manual at next schedule period change”: Can you reproduce this with 1.18-fas?

**State after power loss**
First i could not reproduce that Power Loss Issue, but then i've found a way how to verify that bug/behavior:

After a power outage (5 secs is enough) the device goes **back to the last state (off, manual, scheduler) which was set using the buttons on the devices**.

But if i have changed mode by MQTT (HASS)  and not using the device-buttons then the that state gets ignored.

It seems the device (MCU) only saves the state after changes made using the buttons. Seems to be a Bug in MCU (which is still closed source from manufacturer).

Can you confirm that behavior?

WThermostat only save the config to its Flash when saving a new config. All other values are read from MCU (current temperature, current state, etc.) or are read from NTP-Server. 
HASS also does not saves the state or tries to fix the state (what if WiFi was down and you have changed state manually. Which state is now right? The last HASS state or the state the thermostat reports).

If you power up WThermostat my Code reads state from MCU and reports to MQTT/HASS.
There is also no "save state" command i can send to MCU.
After power up the device Hand or Clock Icon instantly appears on the display, even before the temperature gets displayed.

Hmmmm. Solution would be to save state ourselves in flash and override at startup. 

Workaround is to change state manually on thermostat to Schedule-Mode and this will appear after a power loss.
Having the option "Switch back to Auto mode from manual at next schedule period change" enabled switches to scheduler at next schedule interval - but only if that last MCU state was _not_ OFF, then it stays off.

Kind Regards
Folke

dmerlani, 2020-11-01:
> Inviato da Libero Mail per iOS

>Hi, on my thermostats it seems that power outage always put them on manual mode 20c. See the video attached. No matter the previous state was put through MQTT or by pressing the buttons.
>The workaround to put immediately the thermostat to auto mode after the power outage would be nice to have.
>Just let me know if you need more tests from me.
>BR

https://youtu.be/yszYzC98FQs

dmerlani, 2020-11-01:
https://youtu.be/yszYzC98FQs
No matter the previous state, no matter how it was set (by MQTT or by buttons), after power outage it goes to manual mode 20C.

blackscreener, 2020-11-01:
@fashberg 
You are right, for me the device goes back to the last state (off, manual, scheduler) which was set using the buttons on the devices.

chipmoca, 2020-11-02:
> Workaround is to change state manually on thermostat to Schedule-Mode and this will appear after a power loss.

I do not use the Home Assistant yet (I plan to do it in the future). Now I control only the buttons. On firmware 1.16b6 it works fine, and on all subsequent ones including 1.19betta1 this bug is repeated.

dmerlani, 2020-11-11:
@fashberg what about this "dirty" thing?
Under the loop function in WBecaDevice.h, I have found the function mcuInitialize that is run just one time.
So I just force the schedule mode off and switch off the thermostat. 
So after a power outage, it force to switch off the thermostat (safer mode).
I would be nice to memorize the "previous" state in the flash and recover it after the power outage but I don't really know how to do it...

if (!isMcuInitialized()){
			mcuInitialize();
            schedulesMode->setString(SCHEDULES_MODE_OFF);
            unsigned char deviceOnCommand[] = { 0x55, 0xAA, 0x00, 0x06, 0x00, 0x05,
        	                                    0x04, 0x04, 0x00, 0x01, 0x00};
        	commandCharsToSerial(11, deviceOnCommand);

hannes684, 2020-12-27:
Even the schedules goes back in default when power is reconnected after power loss.
And 15Grad Celcius is to low for the girls in my house :)
I hope this will solve in the future.
That schedules can be saved as default.

blackscreener, 2022-10-02:
So what about this issue? Also schedules goes back to default after power loss.

closed 2023-01-07 by github-actions[bot]