fashberg/WThermostatBeca #60: Self-heating

tshcherban, 2020-10-24:
Im facing self heating problem, both wifi and non-wifi versions of _dry-contact_ thermostats are warmer by 1-2 degrees than ambient, also reporting temperature 1.5-2 degrees more than than two temp sensors (xiaomi and inkbird bluetooth version). Do you know any hardware hack like additional ventilation holes or something other? I can set temp offset but i dont know if its constant in all temperature range.
Here is a photo of wall with thermostat (captured by seek thermal
 compact pro:
[image]

odelma, 2020-11-29:
I've noticed the same and there's been some discussion about this under another issue: https://github.com/fashberg/WThermostatBeca/issues/68#issuecomment-730299913
https://github.com/fashberg/WThermostatBeca/issues/68#issuecomment-735358959

It would make sense to continue the discussion here in case we can come with either a harware or firmware modification to take care of it.

tshcherban, 2020-11-30:
@odelma 
I've looked with an infrared camera, the most warm unit is proprietary MCU and the LCD controller on the board (and ESP is quite hot in WiFi version). But all of theese are quite stable over room temperature range (17..24 degrees), and self-heating becomes faster when relay is on (2w thermal actuator is attached, so current is below 10 mA in sustained mode), seems that circuit dedicated to relay control also heats up, or MCU is heating faster due to increased current over some of its inputs.
All tests i've made with LCD-buttons sandwitch detached from the mounting box, to exclude AC-DC power supply and relay thermal influence.
One of the possible workarounds would be increasing the offset while relay is on, but i'm not sure that we can set this parameter from ESP to tuya MCU.
Another option is using external temperature probe or moving existing thermistor outside of thermostat case a bit, but it would look not very good.
Currently i've done some ugly 'case hack' (holes on top side, removed metal shield on the bottom), but have doubts that it would help.
[image]
[image]

In few weeks i'm planning to remove the power supply and relay completely, feed the thermostat with 12v via two wires directly, and use thermostat only as control panel and temperature probe, and the heating control would be done via some home automation system (currently i have Home Assistant), which reads current temperature and setpoint from the thermostat (but this requires WiFi or modbus version).

paveq, 2020-11-30:
I have also issue with the unit(s) heating internal probe when heating cycle is on. What I did is change deadzone temperature setting from 1 to 2 degrees, and temperature compensation from -3 to -4 degrees. Now heating cycles are bit more stable and longer at least.

Seems when changing deadzone the unit then starts heating only when measured temp goes 2 degrees below the setpoint, and turns off then setpoint is reached. This is why changing compensation is needed (or alternatively set the setpoint up much higher).

odelma, 2020-11-30:
I need to test a little bit the compensation and deadzone settings. I think they are not near optimal right now. Also I need to create a virtual sensor of the floor temperature as I have probably a 15k from the legacy installation and Moes is configure with 10k.

But if nothing else works, I plan to add an extra Xiaomi sensor to bathroom and create an automation that is based on the temperature on the Xiaomi sensor. With a help of a user variable I should be able to still use the thermostat for controlling the "true" setpoint even if the automation would need to set it after that higher to make sure it stays on until Xiaomi sensor reaches the desired temp. But it would be best if this could be avoided with adjusting just the thermostat settings...👍

paveq, 2020-12-01:
I'm thinking it might be best option to use floor sensor in the thermostat itself, and control the setpoint using external temperature sensor such as RuuviTag. Such control loop could be based on PID controller with quite a long integral time.

@odelma are you sure you cannot change the floor sensor? Usually they are placed in a conduit and can be pulled out relatively easily. Maybe yours is stuck (in concrete?) and cannot be pulled out, but you could still push second NTC into the conduit next to previous one?

odelma, 2020-12-01:
> are you sure you cannot change the floor sensor? Usually they are placed in a conduit and can be pulled out relatively easily. Maybe yours is stuck (in concrete?) and cannot be pulled out, but you could still push second NTC into the conduit next to previous one?

Will need to check this - it might be possible. At least the option to try to slide the included sensor there as close as possible to the floor. On the other hand, if I've understood correct the floow sensor is there only for the overheat protection. So if I calculate what is the equivalent of e.g. 30C with the current NTC and then use that as the overheat protection value (and maybe create a virtual sensor in my HA system just that I can check the true floor temperature), it would do the job as well...?

paveq, 2020-12-01:
@odelma Yes, that would also work. Use of the floor sensor depends on "sensor selection" choice you've made in the thermostat: https://github.com/fashberg/WThermostatBeca/blob/master/docs/BHT-002-Manual.pdf

closed 2022-11-05 by github-actions[bot]