fashberg/WThermostatBeca #63: Unknown commands on BHT-002-GCLW

jkrivas, 2020-10-25:
Hello,

I flashed the BHT-002-GCLW with the firmware and tried to use it with Home Assistant. While the auto-discovery worked, every value in HA showed _unknown_.

After some debugging I noticed that MQTT is not being updated with thermostat stats, only network, clock and logging properties are there.

Using Live WebLog I noticed some errors:
```
warning: Not sending state via MQTT home/bedroom/stat/things/thermostat/properties, deviceStateComplete=false
```
and 
```
warning: MCU: unknown;55 aa 03 07 00 05 13 04 00 01 00
warning: MCU: unknown;55 aa 03 07 00 05 07 01 00 01 00
warning: MCU: unknown;55 aa 03 07 00 05 0b 04 00 01 00
warning: MCU: unknown;55 aa 03 07 00 08 0c 02 00 04 00 00 00 00
warning: MCU: unknown;55 aa 03 07 00 05 0d 05 00 01 00
warning: MCU: unknown;55 aa 03 07 00 05 0e 04 00 01 00
```

Checking the source code it seems that the BHT-002-GCLW is using different commands. I was able to figure out that the 
```
warning: MCU: unknown;55 aa 03 07 00 05 0e 04 00 01 00
```
is the lock status. I am not sure about the other commands.

fashberg, 2020-11-01:
Hi @jkrivas,

can you please check if you have same model from DLX like issue #68 ?
Otherwise: please specify the hardware.

Kind Regards
Folke

jkrivas, 2020-11-18:
Sorry about the long time to respond, but I can confirm that the board is exactly the same as in issue #68. Closing the issue

closed 2020-11-18 by jkrivas