fashberg/WThermostatBeca #65: Wrong state in Home Assistant

blackscreener, 2020-10-26:
When thermostat is in auto mode (scheduler) and I change temperature in Home Assistant then the mode in HA changes to manual but mode in device stays on scheduler. So state in device and state in Home Assistant is different,

fashberg, 2020-11-01:
Hey @blackscreener,

which version do you have installed on your thermostat?
Should be fixed with 1.17-fas

blackscreener, 2020-11-01:
> 
> 
> Hey @blackscreener,
> 
> which version do you have installed on your thermostat?
> Should be fixed with 1.17-fas
Hey.
1.18

dmerlani, 2020-11-10:
Hi guys, I belive I have found the solution
changed the onChangeTargetTemperature function adding the "heat" mode command 
unsigned char deviceOnCommand[] = { 0x55, 0xAA, 0x00, 0x06, 0x00, 0x05,
        	                                    0x04, 0x04, 0x00, 0x01, 0x01};
        	commandCharsToSerial(11, deviceOnCommand);

so everytime the user changes the temperature in auto mode, it will force first to switch to heat mode and then change the temperature.

@fashberg Is it correct? can you implement the change?

void onChangeTargetTemperature(WProperty* property) {
		network->log()->trace(PSTR("Got new targetTemperature"));
		bool force=false;
		if (schedulesMode->equalsString(SCHEDULES_MODE_AUTO)){
			network->log()->trace(PSTR("Got new targetTemperature: Switching from Schdule to Manual"));
			schedulesMode->setSuppressOnChange();
			schedulesMode->setString(SCHEDULES_MODE_OFF);
			force=true;
            unsigned char deviceOnCommand[] = { 0x55, 0xAA, 0x00, 0x06, 0x00, 0x05,
        	                                    0x04, 0x04, 0x00, 0x01, 0x01};
        	commandCharsToSerial(11, deviceOnCommand);
		}
		if (schedulesMode->equalsString(SCHEDULES_MODE_OFF)){
			// only set targetTemperatureManualMode and targetTemperatureManualModeToMcu() if current mode is Manual
			if (force || !WProperty::isEqual(targetTemperatureManualMode, this->targetTemperature->getDouble(), 0.01)) {
				targetTemperatureManualMode = this->targetTemperature->getDouble();
				network->log()->trace((String(PSTR("onChangeTargetTemperature, temp: "))+String(targetTemperatureManualMode)).c_str());
				targetTemperatureManualModeToMcu();
				schedulesMode->setString(SCHEDULES_MODE_OFF);
			} else {
				network->log()->trace((String(PSTR("setTargetTemperatureNoChange, temp: "))+String(this->targetTemperature->getDouble())).c_str());
			}
		}
		updateRelaySimulation();
    }

Tigger2014, 2020-12-23:
Hi @fashberg,

I'm seeing the same thing when asking Google assistant via home assistant to turn up the temperature, if in auto mode it does not change to manual, is the solution above the correct way to solve it?.

Many thanks

l0cal-h05t, 2022-02-25:
hi i have the same issue, 
but if i try to compile a new version i get the following error: 
WThermostat\../lib/WAdapter/WAdapter/WJsonParser.h: In member function 'WProperty* WJsonParser::endTrue()':
WThermostat\../lib/WAdapter/WAdapter/WJsonParser.h:500:23: error: cannot convert 'bool' to 'WProperty*' in initialization

can some one help me ?

mromaithi, 2022-08-02:
Hello guys,

Any updates on this bug.

Thanks.

closed 2022-11-07 by github-actions[bot]

labodj, 2026-11-08:
I maintain a fork of this project and have released [v1.26.0-labo](https://github.com/labodj/WThermostatBeca-labo/releases/tag/v1.26.0-labo). It fixes the missing manual-mode command when changing the target from Home Assistant/MQTT while scheduling is active. The requested temperature is saved first, then the firmware sends the manual-mode frame followed by the new setpoint frame. This keeps the ESP's reported mode and the command sent to the thermostat MCU consistent.