fashberg/WThermostatBeca #68: A new panel has appeared for the BHT-002 thermostat.

kocsishome, 2020-10-30:
A new panel has appeared for the BHT-002 thermostat. The measured temperature on the web interface is incorrect. The value displayed is exactly half the value indicated on the web interface.
MQTT values are also bad!

Is there a solution?
[image]
[image]

kocsishome, 2020-10-31:
[image]

fashberg, 2020-11-01:
Hey @kocsishome ,

ok, completely new board.

Can you please

* gimme a link where this version is available to buy
* open WebLog, set WebLog-Level to Verbose and then to a reinitialization (not reboot, reinit)
* Can you set temperature to quarter degrees, or is temperature display more acurate than have degrees?

Kind Regards
Folke

alessandrosalafia, 2020-11-01:
Hi Folk,
I bought the thermostat on the official Mouse store, aliexpress. I believe they will now use this hardware for the next thermostats as well.
https://moes.aliexpress.com/store/4988128?spm=a2g0s.9042311.0.0.2ec64c4dRW7fOK
I have also attached weblogs with verbose mode. If you need other weblogs let me know.

The thermostat shows the setpoint temperature double that displayed on HA, while the measured temperature is wrong. Seems like a mqtt sync problem. Every now and then it disconnects and I see many errors in the log.

kocsishome, 2020-11-02:
Hi!

I bought in the Official Store!

[](https://www.aliexpress.com/item/32984963838.html?spm=a2g0s.9042311.0.0.5adc4c4d7zx08Z
)
I uploaded the diary. I also uploaded an example image.

Thank you for helping!

[image]
[BHT-002_new panel_log.txt](https://github.com/fashberg/WThermostatBeca/files/5476370/BHT-002_new.panel_log.txt)

kocsishome, 2020-11-03:
I discovered another error!

The product cannot display and handle 0.5 values!
[image]
[image]

mVik85, 2020-11-04:
Hello!
Same issue here, opened a dispute.
Checked the user review under the product listing, according to the pictures some have 0.5 values, some doesn't.

Pretty surprising downgrading...

alessandrosalafia, 2020-11-04:
> Hello!
> Same issue here, opened a dispute.
> Checked the user review under the product listing, according to the pictures some have 0.5 values, some doesn't.
> 
> Pretty surprising downgrade...

I have seen that my thermostat is also not accurate to 0.5. Have you opened a dispute and are sending him back?

mVik85, 2020-11-04:
Well, sending anything back to China would not make sense, I asked for a refund, or a technical solution.
Smart Life app shows the following versions:
Main Module: 3.1.1
MCU Module: 1.0.0

There might be an update coming...
Until that I have to look for an other cheap DIY friendly wifi thermostat. Any ideas?

mVik85, 2020-11-04:
I get this decoded with Domoticz Tuya plugin:
Loaded: {'1': True, '2': 23, '3': 24, '4': 'manual', '7': False, '11': '0', '12': 0, '13': 0, '14': 'warming', '19': 'c'}

Seems to have a different dps list.

alessandrosalafia, 2020-11-04:
> Well, sending anything back to China would not make sense, I asked for a refund, or a technical solution.
>
> Smart Life app shows the following versions:
>
> Main Module: 3.1.1
>
> MCU Module: 1.0.0 [...]

Unfortunately no idea.  Older thermostats worked fine, other cheap 8266 based versions there would be different, like Boek, Floueron

alessandrosalafia, 2020-11-05:
I managed to get new thermostats sent to me but asked for the old ones not the DLXs. Today I tried to integrate the thermostat with the Flouerton component or with tuya integration, but with the first it doesn't work and with the second if you turn off the thermostat then it is lost, it is not available. Bad system.

mVik85, 2020-11-05:
> I managed to get new thermostats sent to me but asked for the old ones not the DLXs.

How did you manage that? Free replacement?
This would be also my goal, but no reaction to my dispute yet...

alessandrosalafia, 2020-11-05:
> > I managed to get new thermostats sent to me but asked for the old ones not the DLXs.
> 
> How did you manage that? Free replacement?
> This would be also my goal, but no reaction to my dispute yet...
> 
> 

Yep.. [image] I was amazed too.

kocsishome, 2020-11-07:
New information:

`warning: MCU: unknown;55 aa 03 07 00 74 11 00 00 70 01 68 00 c8 02 b2 00 96 03 2a 00 96 04 1a 00 c8 01 68 00 c8 02 b2 00 96 03 2a 00 96 04 1a 00 c8 01 68 00 c8 02 b2 00 96 03 2a 00 96 04 1a 00 c8 01 68 00 c8 02 b2 00 96 03 2a 00 96 04 1a 00 c8 01 68 00 c8 02 b2 0`

mVik85, 2020-11-11:
I could make it more or less work, with the Domoticz Tuya python plugin:
https://github.com/iasmanis/Domoticz-Tuya-Thermostat-Plugin

It has a little different dps config:

1: True / False - on/off state
2: 26 - setpoint
3: 25 - sensor temperature
4': 'manual' / 'week' / 'eco' - mode
7: True / False - lock status
11: '0' - ?
12: '0' - ?
13: '0' - ?
14: 'warming' / 'heating' - relay state
19: 'c' - ?

The relay state does not get updated consistently, but this could be related to my python skills...

alessandrosalafia, 2020-11-12:
Interesting.  I would like to know what Folke thinks and if he has understood why it works badly.  Is it the fault of the hardware or must the software be adapted?

mVik85, 2020-11-12:
I don't think the device is not okay, I just quickly hacked the python script for Domoticz with limited knowledge, so maybe I missed something, but this is not relevant for this topic:)
Just to confirm, do you also have to raise the setpoint +2°C above the actual temperature to make the relay pulled? (deadzone: 1°C) It is disturbing, if I want heating @23°C room temperature I have to set the setpoint minimum to 25°C and it switches off when 24°C is reached. I don't think hysteresis should work like this...
Btw. I'm also getting the replacement part from the Official Moes store, this is what I call customer orientation!

alessandrosalafia, 2020-11-12:
> I don't think the device is not okay, I just quickly hacked the python script for Domoticz with limited knowledge, so maybe I missed something, but this is not relevant for this topic:)
>
> Just to [...]

I confirm that even when checking it from your Tuya app it is very inaccurate.  High hysteresis level, +1 setpoint tolerance, 3 degrees lower than the actual measured temperature.  setpoint must be raised plus 2 degrees to start the relay.  It sucks a bit.

odelma, 2020-11-12:
I just got the first unit and it's with the fimware that does not support tuya convert, but still with the "normal" hardware. I connected it to Domoticz with the plugin you linked and can confirm that it requires about 1,5-2C higher setpoint than the ambient temperature to switch on. I wonder why it is designed like this. I have not yet tested when it would switch off, but I guess it is 1C below the setpoint?

Does it work like this with the custom firmware as well? If not, I guess I need to start soldering... 😃

realthk, 2020-11-13:
That is how it works, yes. This thermostat switches on 1C below, and off when reaching the setpoint. Which means if you want to force it on, you have to set at least 0.5C above current room temperature indeed. So hysteresis is shifted by +0.5C, otherwise it's the same 1C fluctuation (+some overshoot because of the still warm radiators) as with other thermostats using setpoint +-0.5C.

alessandrosalafia, 2020-11-13:
> That is how it works, yes. This thermostat switches on 1C below, and off when reaching the setpoint. Which means if you want to force it on, you have to set at least 0.5C above current room [...]

In theory they should work like this. The old models are good but these new ones are just bad. Mine detect 3 degrees less than the actual temperature, they do not have half a degree of difference so if you want to increase the temperature you have to raise the setpoint beyond the normal because if it marks 20 degrees, maybe it's 20.5 which then the reality could be almost 21. So you never know what the real temperature is. I verified that with 19 degrees actually displayed with the fireplace on I am 22. I don't know cloud usually, but the old tuya integration on Home assistant worked. This no. If I turn off the thermostat then I don't turn it on anymore from HA, lost. Mouse Official says they are faulty and they sent me more. I specifically asked for the old bac-002 20190409 models. Hopefully good.

paveq, 2020-11-19:
In my tests (old model) when thermostat relay is on, it causes thermostat unit's temperature meter itself to heat up to 2-3 degrees, which seemingly is not compensated in the thermostat firmware itself in any way. This causes heating periods be quite short, and the floor does not even heat up a degree during one cycle due to amount of concrete mass.

I'm using electric floor heating, 230VAC with around 1.5-2 kW of power cables. In nordic countries this level of power seems quite usual, but perhaps these units are designed for much lower power heating cables? Pulling this power through the relay will cause some heat loss.... has anyone tried improving ventilation of these units?

At the same time, I suppose this mechanism by accident also prevents overshooting, which would be quite significant otherwise.

Perhaps we could compensate this by increasing target temperature when heating cycle is on, although then display would come on twice per heating cycle?

odelma, 2020-11-29:
> In my tests (old model) when thermostat relay is on, it causes unit's room temperature probe itself to heat up to 2-3 degrees, which seemingly is not compensated in the thermostat firmware. This causes heating periods be quite short, and the floor does not even heat up a degree during one cycle due to amount of concrete mass contained in the floor.

This is probably wtong issue to discuss this, but I’ve noticed the same (also in a nordic country). If the setpoint is close to ambient temperature it looks like switching on causes fast heating of the unit and this result into on/off type of flickering action. Especially during night time the sound caused by this is not very nice. And probably it does not do good for the unit itself. Have you come up with any workarounds? Thw unit is inside tile wall in quite tight space, some extra ventilation might help. Or a surface mount box. Or external temp sensor which would be used to set the setpoint 2-3C higher than real requested temp, then back to lower once it is reached...

klausahrenberg, 2021-01-23:
> It has a little different dps config:
> 
> 1: True / False - on/off state
> 2: 26 - setpoint
> 3: 25 - sensor temperature
> 4': 'manual' / 'week' / 'eco' - mode
> 7: True / False - lock status
> 11: '0' - ?
> 12: '0' - ?
> 13: '0' - ?
> 14: 'warming' / 'heating' - relay state
> 19: 'c' - ?

If someone is interested to test. I implemented the support for DLX_LH01 in my firmware according to the list above. in my version 1.20i. Needs further testing and 'warming'/'heating' not impl. yet. 
https://github.com/klausahrenberg/WThermostatBeca

closed 2022-11-02 by github-actions[bot]