fashberg/WThermostatBeca #69: WthermostatBeca New Version DLX Control Solutions

alessandrosalafia, 2020-10-31:
Hi Folk . I recently ordered new home thermostats. It arrived and I found they have different hardware. It's still 8266 and I was able to load your firmware, but there are problems in displaying and exchanging data.  
It seems that the thermostat is not communicating correctly with HA. See and set different temperatures. I used mqtt and autodiscovery, but doesn't work better. The firmware is WThermostat 1.18-fas, but  I also tried with 1.17.
[image]
[log new Thermostat.txt](https://github.com/klausahrenberg/WThermostatBeca/files/5468789/log.new.Thermostat.txt)

fashberg, 2020-11-01:
Hey @alessandrosalafia 

i will merge with #68 

Can you also gimme a hint where you have bought?

Kind Regards
Folke

closed 2020-11-01 by fashberg

alessandrosalafia, 2020-11-01:
Hi Folk, 
I bought them on the official Mouse store, aliexpress. I believe they will now use this hardware for the next thermostats as well.
https://moes.aliexpress.com/store/4988128?spm=a2g0s.9042311.0.0.2ec64c4dRW7fOK
I have also attached weblogs with verbose mode. If you need other weblogs let me know.