fashberg/WThermostatBeca #71: Setting of static IP possible?

MichiW1, 2020-11-01:
Hi all,
what a great project here, I like it a lot, thanks for your effort, fashberg!
I am wondering if I can configure a fixed IP address to the device. On tasmota I do it via the Build-in console, but here I have no idea.
I do not like DHCP for my fixed SmartHome device structure, sorry ;-)
Thanks for your help,
Michael

kocsishome, 2020-11-02:
Helló!

Set it up on the network device (router).

MichiW1, 2020-11-02:
Thank you, yes I did so already... however, by certain reasons I‘d like to completely avoid using DHCP. Maybe there is a way, but for the moment I can work with DHCP and fixed assignment in router.

wongcz, 2020-11-10:
Please consider ability to set fixed IP address. I have already set reserved IP on DHCP server.

wongcz, 2020-11-10:
Please consider ability to set fixed IP address. 
I already have reserved IP on DHCP server, but if for some reason thermostat is booted before DHCP server the IP address is not leased.
I have two thermostats and both behave the same (I have latest FW 1.19.beta1, but the same behaviour occured with 1.17 and 1.18)

Log from ISC-DHCP server on linux:
Nov 10 09:53:38 localhost dhcpd[3328]: DHCPREQUEST for 0.0.0.0 from a4:cf:12:e6:c1:5a via enp3s0: unknown lease 0.0.0.0.
Nov 10 09:53:38 localhost dhcpd[3328]: DHCPREQUEST for 0.0.0.0 from d8:bf:c0:0c:d1:4d via enp3s0: unknown lease 0.0.0.0.

Google says this is behaviour of "faulty" DHCP client.

The only way to get it working is to physically disconnect and reconnect thermostat (the ESP part with display).

closed 2022-11-01 by github-actions[bot]