fashberg/WThermostatBeca #76: problem at startup

mcike59295, 2020-11-23:
if I restart the home assistant or a wifi cut, the thermostat returns to the temperature of 10.5!
is this normal?
cordially

really good work 👍

rumoldus, 2020-11-24:
Same thing here. Somehow the thermostat very easily falls back to manual mode. The setting to revert back to schedule helps, except when HA is down. And it should not be necessary in the first place, power dips do not occur here. Not sure whether this is ESP related or MCU. Some more logging (history) could help here.
Super project!

dmerlani, 2020-11-24:
Hi, I have modified the beta 19 in order to start in OFF mode after power outage. Waiting for Folke to solve this issue on the official release.
[wthermostat-1.19.beta1-merlo.bin.zip](https://github.com/fashberg/WThermostatBeca/files/5590650/wthermostat-1.19.beta1-merlo.bin.zip)

mcike59295, 2020-11-24:
Merci beaucoup 😊

dmerlani, 2020-11-24:
Please try this firmware and let me know!
Also, when you are in schedule mode and you change the temperature, it correctly changes to manual mode setting the temperature 
BR

Inviato da Libero Mail per iOS

mcike59295, 2020-11-24:
I installed the firmware, when I restart the home assistant server the temperature does not stay in memory and always drops back to 10.5

closed 2020-11-25 by mcike59295