fashberg/WThermostatBeca #78: Random restart

erelor, 2020-11-25:
I flashed 3 different BECA BHT-002 with 1.18 fas firmware and all of these randomly restart about once per day. I noticed it looking at Info/Uptime information. 
The real big trouble is that when the device restart it goes in Eco Mode and the target temperature is set to 20 °C.

rumoldus, 2020-11-25:
I suspect I have the same thing. In Issue #76, I mentioned that my thermostat falls back to manual mode regularly (once a day could be about right). When the flag to resume the schedule is active, it usually recovers, but I also had an instance where Home Assistance had crashed and the thermostat went to Eco mode and statyed there. Freaking cold bathroom floor.

svenappel, 2020-11-30:
I have exactly the same problem. My BHT-002 restarts at least once a day and it then will set the target temperature to 20 degrees wasting energy.

kovacsm, 2021-01-24:
I have realized, my thermostat restartes itself, when cannot connect to the NTP server, and start with manual mode...

closed 2022-10-31 by github-actions[bot]

poseidogithub, 2023-10-30:
just openes a new thread about this issue, in my case the restart is each 4 to 5 minutoes, on my 3 thermostats:

https://github.com/fashberg/WThermostatBeca/issues/211