fashberg/WThermostatBeca #79: Change temperature sensor

dimvia, 2020-11-25:
is it possible to adjust to another temperature sensor? for example 6.8 ntc?

gavlexx, 2020-11-28:
Also want to know is is possible to adjust thermostat to another NTC.
Changed thermostat from Danfoss DEVI ECtemp Next Plus with NTC 15k. Cannot change NTC cause it is in floor.
Anybody knows how to convert between NTC?

odelma, 2020-11-29:
I have the same issue. Original was in place and it seems to shows half of the real floor temperature. I wonder if I can do anything about it?

gavlexx, 2020-11-29:
> I have the same issue. Original was in place and it seems to show half of the real floor temperature. I wonder if I can do anything about it?

Not half. It simply calculate temperature for 15k as it was 10k sensor.
Say, it shows you floor temperature 15C, so it measures resistance, corresponding 15C for 10k sensor. But it is really 15k sensor, so for this resistance fits 24-24,5C (see table below) 

[image]

dawidm82, 2020-11-30:
I changed the R15 resistor to match the thermistor. The default is 10k (103, 01C), if you have 6.8k thermistor change it to 6.8k. This works because the ratio is measured.

odelma, 2020-11-30:
Can elaborate a bit, please? I.e. What needs to be done in practice - and how to do it? 😄

gavlexx, 2020-11-30:
> I changed the R15 resistor to match the thermistor. The default is 10k (103, 01C), if you have 6.8k thermistor change it to 6.8k. This works because the ratio is measured.

I cannot change resistor cause it is in floor forever. Unfortunately my builders didn't make it changable. So I should do calculations in HA to obtain real floor temperature.

It seems rather strange not to have options for different resistors in Moes thermostat...

gavlexx, 2020-11-30:
> I changed the R15 resistor to match the thermistor. The default is 10k (103, 01C), if you have 6.8k thermistor change it to 6.8k. This works because the ratio is measured.

Do you mean you should only change resistor on thermostat motherboard? 
Don't this affect another thermistor for air?

gavlexx, 2020-11-30:
@dawidm82 , is this resistor used for measuring floor temperature? 
(one picture for BAC, another - for BHT)

[image]
[image]

dawidm82, 2020-11-30:
I had the same problem. Because MCU measures the temperature, it cannot be adjusted in the software. You have to modify the potential divider in the thermostat so that the MCU gets the same voltage levels with another thermistor in the floor.
This is really simple but requires some soldering skills. You need to replace R15 (part of the voltage divider) on the thermostat PCB to keep the ratio (1:1) the same. For example, for a 100k thermistor, it will be a 100k resistor.
This will not affect the temperature sensor, it has a separate divider (R16).

dawidm82, 2020-11-30:
Yes that's this one.

D-side, 2021-04-11:
> EDIT: sorry, not sure about R19, should be connected to the sensor on one side, and to 5V on the other

Probed R19 on the same board with a multimeter. One side is connected to 3rd pin from the right in the white connector, which leads to one of the sensor terminals, Not sure how to find 5V quickly, but I found a contact with the power pin on the flash chip right above the white connector, which accepts up to 5.5V according to the datasheet; and a whole bunch of other points too, including top of C17 on the right. Feels like a decent guess, especially since no other 01C on the board behaves like that. 🤷

I'll give that a try with 6.8k with an NTC 6.8k already in my floor and report back how it went. 👍

_(If a lot of time has passed and I haven't reported back still, feel free to nudge me)_

Eriond, 2021-08-08:
Thanks for the tip about the voltage divider.
This is what I found in my device, an AVATTO such as the one linked from the [Tasmota/Blakadder site](https://templates.blakadder.com/ME81H.html):
[image]

martni, 2022-02-14:
Hello,
I have same problem with floor heating sensor. I can't change it. But before I get it know I bough from Aliexpress from MincoHome thermostat what is using 10KOhm external sensor, but in my bath floor is installed 47KOhm sensor. 
Link: https://www.aliexpress.com/item/1005003091456125.html?spm=a2g0o.9042311.0.0.2c5d4c4d984Igq

Is somebody already have modified this kind of thermostat? Which resistor is part of sensor voltige driver to change?
[image]
[image]

Regards,
Mart

Eriond, 2022-02-14:
Looking at the pictures you have provided, I'd say R7 is the most likely candidate.
It is coupled using C11 as a low-pass filter. The fractional voltage is then fed via a trace that goes back under JP1 and over to the microcontroller, where it drops down a via near the U6 label.
I see the JP1 header is broken; the lower tab seems to have sustained some excessive force. I hope you can put it together again without the cable connector coming loose from JP1.

martni, 2022-05-02:
Hi Eriond,
You were right. I checked your succession and changed the R7 resistor from 10k to 47k. The thermostat working with 47k NTC. :-)
Now I need to found a way how to control the thermostat with Domoticz.

closed 2022-10-31 by github-actions[bot]

Duracell79, 2022-11-19:
Hi. I have same problem. I bought Meross MTX-200 thermostat and it’s using 10k Ohm external sensor and my floor has 47k Ohm sensor. Can anybody help me to find that resistor what need to be replaced?

[image]
[image]

Duracell79, 2022-11-19:
[image]

Duracell79, 2022-11-19:
[image]

Duracell79, 2022-11-19:
[image]

Duracell79, 2022-11-19:
Can it be R16 it leads to other sensor connector?

InvZim, 2022-11-26:
> Hi. I have same problem. I bought Meross MTS-200 thermostat and it’s using 10k Ohm external sensor and my floor has 47k Ohm sensor. Can anybody help me to find that resistor what need to be replaced?
> 
> [image] [image]

Hi, did you find this out? I'm interested having the same one.

Duracell79, 2022-11-26:
Not yet. In one week i will post the result

mklik, 2022-12-16:
Worked for me with the BAC-002 also for a 100K NTC.
Thanks for the descriptions.

kaar3l, 2024-11-03:
Thank you for the thread and info. I had 15k NTC sensor in floor, but the de
vice needed 10k. So I found one 10k resistor that was connected to external sensor. I added 5k resistor, so it would be 10k+5k=15k. Now it seems to be working. The schematic on my device is a bit different.
R22 is in my case the resistor that needs to be changed: (I also changed the orginal esp like device)
[image]
My device looks like that:
[image]

TurtleFX, 2024-12-16:
I bought a MOES BHT-006 Zigbee Thermostat (BHT-006GBZB) to replace my Devireg Thermostat with an installed 15k NTC sensor. I replaced the R14 10k resistor (which is connected between VDD and one of the NTC sensor nodes) with a 15k resistor, and it works as intended.
[image]