fashberg/WThermostatBeca #89: Thermostat model BHT 8000

grbasta, 2020-12-10:
Hello, in your opinion this firmware, can it be good for this model?
[image]

Master-Rudi, 2020-12-10:
Would be awesome if it will work!

nwaelti, 2021-01-13:
It surely works, as I've based my own firmware on codes found in this repository, I can even send weather infos to it ;)

grbasta, 2021-01-14:
> It surely works, as I've based my own firmware on codes found in this repository, I can even send weather infos to it ;)
> [image]

Hello and so what firmware can we use, to put it locally?
Thanks

nwaelti, 2021-01-14:
You can use this [ fashberg ] firmware on it...
https://github.com/fashberg/WThermostatBeca/releases/tag/v1.18-fas
It will work just fine

grbasta, 2021-01-16:
> You can use this [ fashberg ] firmware on it...
> https://github.com/fashberg/WThermostatBeca/releases/tag/v1.18-fas
> It will work just fine

Hello, but do you need to flash with cables or with tuya-convert? I tried tuya-convert but it doesn't work. Do you have the right procedure? Thanks

nwaelti, 2021-01-16:
Tuya convert never worked for me... I unsoldered the chip (TYWES3) and flashed it outside of the thermostat with a flashing tool and resoldered it afterward in the thermostat.

grbasta, 2021-01-16:
> Tuya convert never worked for me... I unsoldered the chip (TYWES3) and flashed it outside of the thermostat with a flashing tool and resoldered it afterward in the thermostat.

Hello sorry if I always ask you, I flashed the firmware you indicated, but unfortunately it does not work on home assistant, temperatures and states, are we sure it works for the bht 8000? I could try what you have, maybe it works. Thanks

[image]

nwaelti, 2021-01-17:
I am also not using HASS, I can only tell you that the code used in this firmware works with the beca serie 8000 well...
Are you able to connect to it with your webbrowser and directly with MQTT ?
You will not be able to use my firmware, I'm not using something that is well known, I'm developed my own smart home system based on MQTT and some of my config is hard coded into the firmware

grbasta, 2021-01-17:
> I am also not using HASS, I can only tell you that the code used in this firmware works with the beca serie 8000 well...
> Are you able to connect to it with your webbrowser and directly with MQTT ? [...]

Good evening what I wanted to report is that the same firmware, installed on BHT-002-GCLW (Water / Gas Boiler), behaves very well with Home Assistant, while installed on the BHT 8000 does not have the same effect, i.e. it does not expose the set temperature and that measured and the on and off commands do not work. While if you want to use it from its touch it works well. It would seem that MQTT does not interact, yet comparing them the settings are identical, obviously the name and related topic change. Greetings

nwaelti, 2021-01-18:
Maybe you could try to change this function
[image]
in WBecaDevice.h to return always true and recompile you own version...
Did you manage to access the website of the thermostat? (http://IP/things) to check if it answers there ?
MQTT topics are not changing as they are managed by this firmware and not he MCU.

grbasta, 2021-01-18:
> Magari potresti provare a cambiare questa funzione in WBecaDevice.h per restituire sempre true e ricompilare la tua versione ... Sei riuscito ad accedere al sito web del termostato? ( [http: // IP / cose](http://IP/things) ) per verificare se risponde lì? Gli argomenti MQTT non cambiano poiché sono gestiti da questo firmware e non dall'MCU.
> [image]
> 
> [](http://IP/things)

Hi, this is what i see if i go from ip / things browser:
[{"name":"Termostato-PT_Thermostat","@context":"https://iot.mozilla.org/schemas","title":"Termostato-PT_Thermostat","@type":["Thermostat"],"properties":{"switchBackToAuto":{"title":"switch Back from Manual to Auto at next Schedule","type":"boolean","links":[{"href":"/things/thermostat/properties/switchBackToAuto"}]},"temperature":{"title":"Actual","type":"number","readOnly":true,"unit":"celsius","@type":"TemperatureProperty","links":[{"href":"/things/thermostat/properties/temperature"}]},"targetTemperature":{"title":"Target","type":"number","unit":"celsius","multipleOf":0.50,"@type":"TargetTemperatureProperty","links":[{"href":"/things/thermostat/properties/targetTemperature"}]},"deviceOn":{"title":"Power","type":"boolean","links":[{"href":"/things/thermostat/properties/deviceOn"}]},"schedulesMode":{"title":"Schedules","type":"string","enum":["off","auto"],"@type":"ThermostatSchedulesModeProperty","links":[{"href":"/things/thermostat/properties/schedulesMode"}]},"ecoMode":{"title":"Eco","type":"boolean","links":[{"href":"/things/thermostat/properties/ecoMode"}]},"locked":{"title":"Lock","type":"boolean","links":[{"href":"/things/thermostat/properties/locked"}]},"mode":{"title":"Mode","type":"string","enum":["off","auto","heat"],"@type":"ThermostatModeProperty","links":[{"href":"/things/thermostat/properties/mode"}]},"action":{"title":"Action","type":"string","readOnly":true,"enum":["off","heating","idle"],"@type":"ThermostatActionProperty","links":[{"href":"/things/thermostat/properties/action"}]},"state":{"title":"State","type":"string","readOnly":true,"enum":["off","heating","cooling","fan"],"@type":"HeatingCoolingProperty","links":[{"href":"/things/thermostat/properties/state"}]},"mcuId":{"title":"mcuId","type":"string","readOnly":true,"links":[{"href":"/things/thermostat/properties/mcuId"}]}}}]

nwaelti, 2021-01-19:
Io non sono securo, ma penso chè dovrebe provare da collegarti sul link chè se vede nel schema:
[image]
Ai provato di recompilare il firmware con la modifica nella funziona hasDeviceWithHassAutodiscoverSupport() ?

grbasta, 2021-01-19:
> Io non sono securo, ma penso chè dovrebe provare da collegarti sul link chè se vede nel schema:
> [image]
> Ai provato di recompilare il firmware con la modifica nella funziona hasDeviceWithHassAutodiscoverSupport() ?

Good morning, then maybe we didn't understand each other, I have the same firmware, I have installed it both on the Moes and on the BHT 8000 as you can see from the attached photo, the only difference is that for the moes I used the tuya at the time -convert and on the BHT 8000 with cables soldered on the TYWES3. Except that for the Moes on Home Assistant no problem, while the BHT 8000 does not detect the status and temperatures, but accepts the commands.
Accessing from the web: on the MOES and it works fine.
http://192.168.2.18/things/thermostat/properties
I read this:
{"idx": "Thermostat-P1", "ip": "192.168.2.18", "firmware": "1.18-fas", "switchBackToAuto": true, "temperature": 22.50, "targetTemperature": 22.00, " deviceOn ": true," schedulesMode ":" off "," ecoMode ": false," locked ": false," mode ":" heat "," action ":" idle "," state ":" off "," mcuId ":" IAYz2WK1th0cMLmL1.0.0 "}
Accessing from the web: on the BHT 8000 and it works badly.
http://192.168.2.16/things/thermostat/properties
I read this:
{"idx": "Thermostat-PT", "ip": "192.168.2.16", "firmware": "1.18-fas", "switchBackToAuto": true, "temperature": 23.00, "targetTemperature": 22.00, " deviceOn ": true," schedulesMode ":" off "," ecoMode ": false," locked ": false," mode ":" heat "," action ":" idle "," state ":" off "," mcuId ":" {"p": "IAYz2WK1th0cMLmL", "v": "1.0"}

I noticed that on the second result after mcuID there is a {too many, so I think that's what messes up the reading on Mqtt.
How can this be solved? I don't know what software you can use.
Thanks

[image]

badger707, 2021-02-08:
@grbasta I'm in the same boat, have 2 of these thermostats...
I have tweaked a little bit the code for this model and recompiled from source, finally got it working properly.
Issue was incorrect/new output JSON formatting as you saw in ProductID:
`notice: Product ID Answer from MCU (commandLength: 42)
notice: MCU: productId;55 aa 03 01 00 2a 7b 22 70 22 3a 22 49 41 59 7a 32 57 4b 31 74 68 30 63 4d 4c 6d 4c 22 2c 22 76 22 3a 22 31 2e 30 2e 30 22 2c 22 6d 22 3a 30 7d
notice: Product ID: '{"p":"IAYz2WK1th0cMLmL","v":"1.0.0","m":0}'`

So I have made "quick&dirty" fix and seems all works fine:

_else if (receivedCommand[3] == 0x01) {
						// Product ID Answer
						// 55 aa 01 01 00 15
						// HE AD VR CM LENGT 
						network->log()->notice(F("Product ID Answer from MCU (commandLength: %d)"), commandLength);
						logIncomingCommand("productId" , LOG_LEVEL_NOTICE);
						knownCommand=true;
						if (commandLength>=5){
							unsigned int len=0;
							//len = ((byte)receivedCommand[4] <<8) + (byte)receivedCommand[5];	// original line
							len = 16;	// badger707
							// char buf[len+1];	// original line
							char buf[17];	// badger707
							for (unsigned int i=0;i<len;i++){
								//buf[i]=receivedCommand[6+i];	// original line
								buf[i]=receivedCommand[12+i];	// badger707
							}							
							buf[len]=0;
							this->mcuId->setString(buf);
							if (mcuInitializeState==2) mcuInitializeState++;
							network->log()->notice(F("Product ID: '%s'"), buf);
						}
					}_

I know thats not ideal way of doing this but it works, and lets hope @fashberg will twist this and will make integration for next release.

Here is view from http://device_ip/things/thermostat/properties
[image]

You can grab compiled firmware binary file for BHT-8000 from here <https://github.com/badger707/WThermostatBeca/tree/master/BHT-8000GCLW>
All credits belongs to original authors, I just done small tweak to get it working for this model.

In thermostat config web page you'll still see old thermostats listed in drop-down list... Have not updated these, but have them selected as BHT-002-GBLW and works just fine

grbasta, 2021-02-09:
> @grbasta I'm in the same boat, have 2 of these thermostats...
> I have tweaked a little bit the code for this model and recompiled from source, finally got it working properly.
> Issue was incorrect/new output JSON formatting as you saw in ProductID:
> `notice: Product ID Answer from MCU (commandLength: 42) notice: MCU: productId;55 aa 03 01 00 2a 7b 22 70 22 3a 22 49 41 59 7a 32 57 4b 31 74 68 30 63 4d 4c 6d 4c 22 2c 22 76 22 3a 22 31 2e 30 2e 30 22 2c 22 6d 22 3a 30 7d notice: Product ID: '{"p":"IAYz2WK1th0cMLmL","v":"1.0.0","m":0}'`
> 
> So I have made "quick&dirty" fix and seems all works fine:
> 
> _else if (receivedCommand[3] == 0x01) { // Product ID Answer // 55 aa 01 01 00 15 // HE AD VR CM LENGT network->log()->notice(F("Product ID Answer from MCU (commandLength: %d)"), commandLength); logIncomingCommand("productId" , LOG_LEVEL_NOTICE); knownCommand=true; if (commandLength>=5){ unsigned int len=0; //len = ((byte)receivedCommand[4] <<8) + (byte)receivedCommand[5]; // original line len = 16; // badger707 // char buf[len+1]; // original line char buf[17]; // badger707 for (unsigned int i=0;i<len;i++){ //buf[i]=receivedCommand[6+i]; // original line buf[i]=receivedCommand[12+i]; // badger707 } buf[len]=0; this->mcuId->setString(buf); if (mcuInitializeState==2) mcuInitializeState++; network->log()->notice(F("Product ID: '%s'"), buf); } }_
> 
> I know thats not ideal way of doing this but it works, and lets hope @fashberg will twist this and will make integration for next release.
> 
> Here is view from http://device_ip/things/thermostat/properties
> [image]
> 
> You can grab compiled firmware binary file for BHT-8000 from here https://github.com/badger707/WThermostatBeca/tree/master/BHT-8000GCLW
> All credits belongs to original authors, I just done small tweak to get it working for this model.
> 
> In thermostat config web page you'll still see old thermostats listed in drop-down list... Have not updated these, but have them selected as BHT-002-GBLW and works just fine
> 
> Edit: for clarity -- above mentioned tweak is a fix for HASS integration (but correct me if I'm wrong).

Good morning in the meantime thank you, in fact it was not read well with mqtt by home assistant and therefore not manageable. When I learn to make the changes myself it will always be too late. I'll test it tonight.
Greetings

grbasta, 2021-02-09:
> @grbasta I'm in the same boat, have 2 of these thermostats...
> I have tweaked a little bit the code for this model and recompiled from source, finally got it working properly.
> Issue was incorrect/new [...]

Hello now in the home assistant I see the temperature of the thermostat that I did not see before, but I don't see the climate entity, does it happen to you too?

badger707, 2021-02-10:
So my initial thoughts was that main issue in HA was about MCUID output format... It seems there is something else incorrect/wrong in payload or its formatting, and HA keeps throwing errors in logs.... 
I'll try to have a look, but I'm no expert in here...

f3nix, 2021-02-10:
Hi.
It looks like temp_step has some trash in it.

```
"temp_step":"1� �",
```
Cheers,
Mateusz

grbasta, 2021-02-10:
> So my initial thoughts was that main issue in HA was about MCUID output format... It seems there is something else incorrect/wrong in payload or its formatting, and HA keeps throwing errors in logs....
> I'll try to have a look, but I'm no expert in here...

However for the moment I have inserted the mqtt platform in the configuration.yaml and I have solved this:
climate:
  - platform: mqtt
    name: Thermostat-P1
    send_if_off: false
    availability_topic: "home / Thermostat-P1 / tele / LWT"
    payload_available: Online
    payload_not_available: Offline
    action_topic: "home / Thermostat-P1 / stat / things / thermostat / properties"
    action_template: "{{value_json.action}}"
    mode_command_topic: "home / Thermostat-P1 / cmnd / things / thermostat / properties / mode"
    mode_state_topic: "home / Thermostat-P1 / stat / things / thermostat / properties"
    mode_state_template: "{{value_json.mode}}"
    away_mode_command_topic: "home / Thermostat-P1 / cmnd / things / thermostat / properties / ecoMode"
    away_mode_state_topic: "home / Thermostat-P1 / stat / things / thermostat / properties"
    away_mode_state_template: "{{value_json.ecoMode}}"
    temperature_command_topic: "home / Thermostat-P1 / cmnd / things / thermostat / properties / targetTemperature"
    temperature_state_topic: "home / Thermostat-P1 / stat / things / thermostat / properties"
    temperature_state_template: "{{value_json.targetTemperature}}"
    current_temperature_topic: "home / Thermostat-P1 / stat / things / thermostat / properties"
    current_temperature_template: "{{value_json.temperature}}"
    payload_on: "true"
    payload_off: "false"
    modes: ["heat", "auto", "off"]
    min_temp: 10
    max_temp: 35
    temp_step: 0.5

badger707, 2021-02-12:
> It surely works, as I've based my own firmware on codes found in this repository, I can even send weather infos to it ;)
> [image]

@nwaelti -- could you bring some light on how you did the trick to send weather info to it? Any pointers?

f3nix, 2021-02-13:
Hi @badger707 .

Here is some info on the weather:
1. [Enabling the function of obtaining weather data (optional)](https://developer.tuya.com/en/docs/iot/device-development/embedded-software-development/mcu-development-access/wifi-mcu-sdk-solution/tuya-cloud-universal-serial-port-access-protocol?id=K9hhi0xxtn9cb#title-19-Enabling%20the%20function%20of%20obtaining%20weather%20data%20(optional))
2. [Sending weather data (optional)](https://developer.tuya.com/en/docs/iot/device-development/embedded-software-development/mcu-development-access/wifi-mcu-sdk-solution/tuya-cloud-universal-serial-port-access-protocol?id=K9hhi0xxtn9cb#title-20-Sending%20weather%20data%20(optional))

I have sent bytes from the example to the thermostat and all the icons appeared.

Cheers,
Mateusz

f3nix, 2021-02-13:
It looks like this thermostat uses newer api for producut information.

It is version 0x03.

[Querying product information](https://developer.tuya.com/en/docs/iot/device-development/embedded-software-development/mcu-development-access/wifi-mcu-sdk-solution/tuya-cloud-universal-serial-port-access-protocol?id=K9hhi0xxtn9cb#title-6-Querying%20product%20information)

This (0x03) version returns in JSON format.

Cheers,
Mateusz

nwaelti, 2021-02-13:
Sure, @badger707 , here's what I'm sending:

For UV/humidity/windlevel:
```
unsigned char wData[54] = {
  0x55, 0xAA, 0x00, 0x21, 0x00, 0x30, 0x01, 0x0A, 0x77, 0x2E, 0x68, 0x75,
  0x6D, 0x69, 0x64, 0x69, 0x74, 0x79, 0x00, 0x04, 0x00, 0x00, 0x00, humidity,
  0x05, 0x77, 0x2E, 0x75, 0x76, 0x69, 0x00, 0x04, 0x00, 0x00, 0x00, uv,
  0x0B, 0x77, 0x2E, 0x77, 0x69, 0x6E, 0x64, 0x4C, 0x65, 0x76, 0x65, 0x6C,
  0x00, 0x04, 0x00, 0x00, 0x00, windlevel
};
```
And the for the weather condition icon:
```
// 0=Sunny, 1=Rainy, ?=Cloudy
	if (condition == 0) {
 		unsigned char rawData[27] = {
			0x55, 0xAA, 0x00, 0x21, 0x00, 0x15, 0x01, 0x0B, 0x77, 0x2E, 0x63, 0x6F,
			0x6E, 0x64, 0x69, 0x74, 0x69, 0x6F, 0x6E, 0x01, 0x06, 0xE6, 0x99, 0xB4,
			0xE4, 0xBA, 0x91
		};
		commandCharsToSerial(27, rawData);
	} else if (condition == 1) {
		unsigned char rawData[27] = {
			0x55, 0xAA, 0x00, 0x21, 0x00, 0x15, 0x01, 0x0B, 0x77, 0x2E, 0x63, 0x6F,
			0x6E, 0x64, 0x69, 0x74, 0x69, 0x6F, 0x6E, 0x01, 0x06, 0xE4, 0xB8, 0xAD,
			0xE9, 0x9B, 0xA8
		};
		commandCharsToSerial(27, rawData);
	} else {
		unsigned char rawData[27] = {
			0x55, 0xAA, 0x00, 0x21, 0x00, 0x15, 0x01, 0x0B, 0x77, 0x2E, 0x63, 0x6F,
			0x6E, 0x64, 0x69, 0x74, 0x69, 0x6F, 0x6E, 0x01, 0x06, 0xE5, 0xA4, 0x9A,
			0xE4, 0xBA, 0x91
		};
```

grbasta, 2021-02-13:
Hi @mwaelti, but how is it implemented? need to change the firmware? I don't have the right skills. A step by step guide? Thank you.

badger707, 2021-02-13:
@grbasta - I think these lines has to be baked into correct place within firmware, and then there must be some side process which is sending/pushing updated forecast information remotely to thermostat... Node Red maybe...? 

 @nwaelti and @f3nix -- big thanks for directions and code! I understand we'd need to send these commands to mcu, but question would be how you're sending all these commands while developing/exploring and before compiling? I have VScode with PlatformIO and I'm able to compile all this from source just fine. However I'm struggling to understand how you all folks sending these testing commands to mcu before compiling so you could capture mcu response and write needed code/functions and then compile.. That  remains a mystery for me.... I'm not a dev or software engineer, just learning...
Any guidance/advise on this?

nwaelti, 2021-02-13:
@badger707 There are actually two ways of seeing things: The first one is to read and understand the Tuya protocol that @f3nix mentioned above, this [link](https://developer.tuya.com/en/docs/iot/device-development/embedded-software-development/mcu-development-access/wifi-mcu-sdk-solution/tuya-cloud-universal-serial-port-access-protocol?id=K9hhi0xxtn9cb#title-20-Sending%20weather%20data%20(optional)) display how to send weather data to the MCU you can see in my first code part above that I'm sending 0x55 0xAA 0x00 0x21 < That one for instance is the command for weather data. In my firmware I created a MQTT function that allowed me to send raw Hex codes to the MCU, so I can directly try several hex codes and receive back what the MCU responded.
The other way, which I did in the begining was to solder a FTDI TTL to the rx and tx pins of the TYWE3S module to listen to what was going on... But really the whole thing is written down in the Tuya Serial Prototol documentation :)

nwaelti, 2021-02-13:
On the other side, I've a NodeJS application that request OpenWeather weather through an API and sends a MQTT command to the thermostat which is then interpreted and converted to a hex command.

nwaelti, 2021-02-13:
@grbasta A step by step guide would be hard to write, You need to be at least able to build your own custom firmware using PlatformIo in order to start with... and as I stated in the begining, I'm not working with the firmware from fashberg at all, I've my own code, I just responded to this thread because it's the only one I found mentionning the BHT-8000.

badger707, 2021-02-13:
@grbasta -- one of ways to compile firmware would be to install Visual Studio Code, then go to Extensions and install PlatformIO IDE. This will install all additional things needed (like C++). Once PIO install finished go to PIO Home page > platforms > embedded > search for "espessif" ans install [Espressif 8266].

[image]
[image]

Then you'd be following original instructions:

`git clone https://github.com/fashberg/WThermostatBeca`
`cd WThermostatBeca`
`git submodule init`
`git submodule update`

Open this (WThermostatBeca) folder with VSCode and your ready to rock.
You'd be editing WBecaDevice.h file
[image]

build versioning is in platformio.ini, look at row 22

Then you'd be compiling with "build"
[image]

Hope this helps.

badger707, 2021-02-13:
@nwaelti -- thanks for notes. Agree, digesting all Tuya protocol would be preferred way to go, I've read through the sections mentioned by @f3nix (thanks!). Overall this makes sense (I think), but learning curve is steep and requires lots of time to understand all bells and whistles for bare minimum results. 
`In my firmware I created a MQTT function that allowed me to send raw Hex codes to the MCU` -- thats really handy! Would you post/share parameters declaration and function code lines from .h file? This would allow me and everyone who is reading this to cut some corners & save some time.... I appreciate your guidance & help, cheers!

badger707, 2021-02-13:
So instructions in this repo shows how to send command to MCU via MQTT
`mosquitto_pub -t "home/test/cmnd/things/thermostat/mcucommand" -m "55 aa 00 1c 00 08 01 14 09 0c 10 30 0f 06"`

Have tried to on/off device and this part goes well:
`mosquitto_pub -h 192.168.44.113 -u "mosquito-user" -P "mosquito-pwd" -t Therm01/cmnd/things/thermostat/properties/deviceOn -m "true"`

Now I'm trying to send command to MCU so I could enable weather function as advised by @f3nix  [1. Enabling the function of obtaining weather data (optional)](https://developer.tuya.com/en/docs/iot/device-development/embedded-software-development/mcu-development-access/wifi-mcu-sdk-solution/tuya-cloud-universal-serial-port-access-protocol?id=K9hhi0xxtn9cb#title-19-Enabling%20the%20function%20of%20obtaining%20weather%20data%20(optional))

Taking payload bytes from that example I'm sending this:
`mosquitto_pub -h 192.168.100.113 -u "mosquito-user" -P "mosquito-pwd" -t "Therm01/cmnd/things/thermostat/mcucommand" -m "55 aa 03 20 00 10 06 77 2e 74 65 6d 70 06 77 2e 70 6d 32 35 08 77 2e 64 61 74 65 2e 37 32"`

... and nothing happens, no response from MCU. What I'm missing?

badger707, 2021-02-13:
Got weather working  -- I was using incorrect payload in earlier comment. Taking payload bytes what @nwaelti have posted did the trick.

for example, SUNNY icon:
`mosquitto_pub -h 192.168.100.113 -d -u "mosquito-user" -P "mosquito-pwd" -t "Therm01/cmnd/things/thermostat/mcucommand" -m "55 AA 00 21 00 15 01 0B 77 2E 63 6F 6E 64 69 74 69 6F 6E 01 06 E6 99 B4 E4 BA 91"`

Seems there is no need in baking anything into firmware, works fine when posting mentioned payloads to xx/mcucommand topic .

Thanks!

grbasta, 2021-02-14:
> Got weather working -- I was using incorrect payload in earlier comment. Taking payload bytes what @nwaelti have posted did the trick.
>
> for example, SUNNY icon:
> `mosquitto_pub -h 192.168.100.113 -d [...]

Hello, explain yourself better, I should just publish this string from mqtt, obviously inserting the thermostat's ip and mosquitto user and password as per your example?
But where do I do it from? sorry thanks

badger707, 2021-02-14:
@grbasta  - sure, I'll try my best: 

Yes you're right, just publishing payload bytes to topic:
- payload bytes (hex):  `55 AA 00 21 00 15 01 0B 77 2E 63 6F 6E 64 69 74 69 6F 6E 01 06 E6 99 B4 E4 BA 91`
- topic: `Therm01/cmnd/things/thermostat/mcucommand`

Here would be high level details of my test setup:
- one VM running Home Assistant with installed Mosquito broker from Add-on Store
- on my local PC (Ubuntu) I push mentioned MQTT payload/commands in terminal window.
- for this, on local PC, I have mosquito client installed. Terminal command:
`root@ubuntusrv:/# apt-get install mosquitto mosquitto-clients`
- Then, on local PC, I'm using these terminal commands to send payload to MQTT broker (home assistant IP) 
`mosquitto_pub -h 192.168.100.113 -d -u "mosquito-user" -P "mosquito-pwd" -t "Therm01/cmnd/things/thermostat/mcucommand" -m "55 AA 00 21 00 15 01 0B 77 2E 63 6F 6E 64 69 74 69 6F 6E 01 06 E6 99 B4 E4 BA 91"`

That should be it, hope this helps.

Your mileage might be different if you're on Windows, but concept is same -- push payload to topic.

I'm planning to use Node Red (HA Add-on) to query weather data and push to thermostats.

grbasta, 2021-02-15:
> @grbasta - sure, I'll try my best:
>
> Yes you're right, just publishing payload bytes to topic:
>
> * payload bytes (hex):  `55 AA 00 21 00 15 01 0B 77 2E 63 6F 6E 64 69 74 69 6F 6E 01 06 E6 99 B4 E4 BA [...]

Thanks, now we just need to understand how to send the right data, it would take something to do it automatically, but here we need the good ones, let's leave time to study for the most prepared on the subject.

badger707, 2021-02-15:
@grbasta -- all needed info is in this topic -- payload bytes posted by @nwaelti (thanks!)
To do this automatically I have installed Node Red Add-on in HomeAssistant.
Initial test configuration would be something like this:
[image]

As you see I have added 3 [Inject] nodes configured for "string" type payload
[image]

Each of them have different payload bytes for sunny/cloudy/rainy icon switching on your thermostat.

These are connected to [MQTT Out] node
[image]

Then you have [MQTT In] node connected to [debug] node
[image]

By injecting sunny/cloudy/rainy static payloads (needed bytes are already provided by @nwaelti ) this shows up in debug window at right side and switching thermostat icon as result.

Overall, full automation flow would look something like this:
- set [inject] node to repeat every 30min
- with node [www request] call OpenWeather API for JSON payload
- pass this JSON to [function] node -- here would be custom JavaScript function (need to write this) to aggregate weather payload (JSON) and it should build correct output payload format for MCU
- send this payload to [MQTT out] node. [MQTT out] node should be configured for topic "xxxxx/cmnd/things/thermostat/mcucommand"

[image]

Hope this helps/clarifies.

badger707, 2021-02-15:
Actually, 2 separate payloads are needed from [function] output:
- one payload for switching "sunny/cloudy/rainy" icons
- other payload for humidity/uv/wind
[image]

nwaelti, 2021-02-15:
Actually, the whole thing can be put together in a single Payload:
You can combine both, you just have to change the 6th Payload (0x30 or 0x15) to match the number of Bytes actually sent
Example:

```
HEADER      Ver    Cmd        Length   Data...
0x55, 0xAA, 0x00, 0x21, 0x00, 0x45, 0x01, 0x0A, 0x77, 0x2E, 0x68, 0x75,
0x6D, 0x69, 0x64, 0x69, 0x74, 0x79, 0x00, 0x04, 0x00, 0x00, 0x00, humidity,
0x05, 0x77, 0x2E, 0x75, 0x76, 0x69, 0x00, 0x04, 0x00, 0x00, 0x00, uv,
0x0B, 0x77, 0x2E, 0x77, 0x69, 0x6E, 0x64, 0x4C, 0x65, 0x76, 0x65, 0x6C,
0x00, 0x04, 0x00, 0x00, 0x00, windlevel
0x01, 0x0B, 0x77, 0x2E, 0x63, 0x6F,
0x6E, 0x64, 0x69, 0x74, 0x69, 0x6F, 0x6E, 0x01, 0x06, 0xE6, 0x99, 0xB4,
0xE4, 0xBA, 0x91
```

badger707, 2021-02-15:
Perfect! Thanks for sharing, that makes sense, cheers!

grbasta, 2021-02-15:
> Perfetto! Grazie per la condivisione, ha senso, applausi!

Potresti condividere il flow completo quando funziona, cosi modifico solo i dati openweathermap con il mio account.
Grazie

badger707, 2021-02-15:
> > Perfetto! Grazie per la condivisione, ha senso, applausi!
> 
> Potresti condividere il flow completo quando funziona, cosi modifico solo i dati openweathermap con il mio account.
> Grazie

I'm afraid my use case is different than yours, I already have weather station/gadget and dont need thermostat to show same.
My plan is to show AirCon/HVAC status on this thermostat -- like "hvac mode" would be reusing "wind speed" icon, then room "air flow" would reuse UV gauge icon, room humidity would go to its own place, "sunny/cloudy/rainy" would show air filter status, etc....
There could be many other use cases...
I try to find time to write this for OpenWeather, but recommendation is to familiarize with JSON and JavaSscript to write & modify per your needs.

grbasta, 2021-02-15:
> I'm afraid my use case is different than yours, I already have weather station/gadget and dont need thermostat to show same.
> My plan is to show AirCon/HVAC status on this thermostat -- like "hvac [...]

In the meantime, thank you for your availability, I'll try to do some testing myself. Unfortunately I am learning and for me who am not a computer scientist, it takes more time. However if you can find some time, and you want to help me, I'll wait for you !!

badger707, 2021-02-15:
> In the meantime, thank you for your availability, I'll try to do some testing myself. Unfortunately I am learning and for me who am not a computer scientist, it takes more time. However if you can [...]

Yes, you'd need some time to learn, but JS is relatively easy language.
Good place to start: [JavaScript Tutorial - w3schools](https://www.w3schools.com/js/default.asp), also [JSON - Introduction - w3schools](https://www.w3schools.com/js/js_json_intro.asp)

grbasta, 2021-02-17:
@nwaelti 

> Actually, the whole thing can be put together in a single Payload:
> You can combine both, you just have to change the 6th Payload (0x30 or 0x15) to match the number of Bytes actually sent
> Example: [...]

Good morning, this is what I pull out of openweathermap:

coord
lon 12.7849
lat 42.1609
weather
0
id 803
main "Clouds"
description "broken clouds"
icon "04d"
base "stations"
main
temp 6.87
feels_like 4.08
temp_min 6.11
temp_max 7.78
pressure 1023
humidity 81
visibility 10000
wind
speed 2.06
deg 30
clouds
all 75
dt 1613548545
sys
type 1
id 6797
country "IT"
sunrise 1613541746
sunset 1613580215
timezone 3600
id 6538684
name "Nerola"
cod 200
  Right now it is cloudy with humidity 81 if the cloud string is 55 AA 00 21 00 15 01 0B 77 2E 63 6F 6E 64 69 74 69 6F 6E 01 06 E5 A4 9A E4 BA 91, how do I pass it the value 81?
in openweathermap uv index is not there so how do you do it?

nwaelti, 2021-02-17:
@grbasta you can implement the "OneCall" API from OpenWeatherMap ([link here](https://openweathermap.org/api/one-call-api)), it brings you the UVi along with the other values.

```
0x55, 0xAA, 0x00, 0x21, 0x00, 0x45, 0x01, 0x0A, 0x77, 0x2E, 0x68, 0x75,
0x6D, 0x69, 0x64, 0x69, 0x74, 0x79, 0x00, 0x04, 0x00, 0x00, 0x00, humidity,
0x05, 0x77, 0x2E, 0x75, 0x76, 0x69, 0x00, 0x04, 0x00, 0x00, 0x00, uv,
0x0B, 0x77, 0x2E, 0x77, 0x69, 0x6E, 0x64, 0x4C, 0x65, 0x76, 0x65, 0x6C,
0x00, 0x04, 0x00, 0x00, 0x00, windlevel
0x01, 0x0B, 0x77, 0x2E, 0x63, 0x6F,
0x6E, 0x64, 0x69, 0x74, 0x69, 0x6F, 0x6E, 0x01, 0x06, 0xE6, 0x99, 0xB4,
0xE4, 0xBA, 0x91
```
humidity has to be passed after the 'w.humidity' value, you can check the example above with for example hxd, and Hex Editor ([here](https://mh-nexus.de/en/hxd/))
[image]
Here you can check that at offset:

- 0x06, value is 0x01 (means success), Ref:
[image]
- 0x07, value is 0x0A because "w.humidity" is 10 bytes long
- 0x08 - 0x11, value represent string "w.humidity"
- 0x12, value is 0x00 (integer)
- 0x13, value is 0x04 (4 bytes long)
- 0x14 - 0x17 value is 0x0 0x0 0x0 0x45, which translate (in Big Endian) to 69%

You can as well send every individual values to the thermostat or stack them together

closed 2022-10-24 by github-actions[bot]

DerDom, 2023-10-19:
> Sure, @badger707 , here's what I'm sending:
> 
> For UV/humidity/windlevel:
> 
> ```

Hey @nwaelti , I bought the BHT 8000 since I wanted a device that can measure the humidity in my bathroom. But it seems that the thermostat can only show humidity that is sent from an external source (but cannot measure humidity). Is my understandig correct?

nwaelti, 2023-10-19:
> Hey @nwaelti , I bought the BHT 8000 since I wanted a device that can measure the humidity in my bathroom. But it seems that the thermostat can only show humidity that is sent from an external source (but cannot measure humidity). Is my understandig correct?
Unfortunately, you are correct, that's exactly how it works :( The device in itself has no humidity sensor

DerDom, 2023-10-19:
> 
> Unfortunately, you are correct, that's exactly how it works :( The device in itself has no humidity sensor

Oh no, it was quite some work to get the device and an esp chip into it. =( 

But thank you for your quick reply! I might use the thermostat to show the humidity of another room, then. =)

@fashberg , could it make sense to add support for BHT 8000, so that setting values becomes easier than publishing payload bytes to MQTT topics?

bordeux, 2024-09-27:
Hey Guys! Thank you for a help. Your comments were very helpful for me :)

I created a code in JS to generate MQTT payload. It works fine for humidity and UV index.

But i have problem with wind speed and icon about weather. What values MCU expecting there?

```javascript

const weatherPayload = function (properties) {
    const stringToHex = (str) => {
        return Array.from(new TextEncoder().encode(str));
    };
    const numberToHex = (num, minBytes) => {
        const byteArray = [];
        while (num > 0) {
            const byte = num & 0xff; // Get last 8 bits (1 byte)
            byteArray.unshift(byte); // Insert byte at the start
            num = num >> 8; // Shift number by 8 bits to the right
        }
        if (byteArray.length === 0) {
            byteArray.push(0);
        }
        while (byteArray.length < (minBytes ?? 0)) {
            byteArray.unshift(0);
        }
        return byteArray;
    };

    let data = [0x01];

    Object.keys(properties).forEach(key => {
        const keyHex = stringToHex(key);
        const value = properties[key];
        const valueType = typeof value;
        data.push(Math.round(keyHex.length));
        data.push(...keyHex);
        let valueHex;
        switch (valueType) {
            case "string":
                data.push(0x01)
                valueHex = stringToHex(value);
                data.push(Math.round(keyHex.length))
                data.push(...valueHex)
                break;
            case "number":
                data.push(0x00)
                valueHex = numberToHex(value, 4);
                data.push(valueHex.length);
                data.push(...valueHex)
                break;
            default:
                throw new Error("Invalid data type value for " + key);
        }
    });

    return [0x55, 0xAA, 0x00, 0x21]
        .concat(numberToHex(data.length, 2))
        .concat(data)
        .map(char => char.toString(16).toLocaleUpperCase().padStart(2, '0')).join(" ")
};

// Example usage
console.log(weatherPayload({
    "w.humidity": 34, //integer, from 0 to 100
    "w.uvi" : 3, // integer, from 1 to 5
    "w.conditionNum" : "120", // string!, codes  available here :  https://gist.github.com/bordeux/f2865fe3e1b94db7b6493af972806555,,
    "w.windLevel": 100 // i dont know
}))
```