fashberg/WThermostatBeca #90: 1.18 Issues in Relay State Calculation

paveq, 2020-12-14:
Issues in relay state calculation  as seen by the chart in 1.18, with these MCU parameters:

- temperature compensation set to -4
- deadband set to 2

[image]

paveq, 2020-12-14:
Weird behaviour I noted with these MCU settings when i turn off the unit, and turn it back again, it starts to heat immediately when target is 22.5 and display shows 22.0. Maybe on startup the unit just behaves in a way, that is measured temp is less than target, it defaults to always heating?

paveq, 2020-12-14:
Maybe I have wrong deadband setting in the thermostat, not sure. Seems no matter what number I input to it in settings, after refresh it always shows 1, and not the value that was stored (if it was stored?).

richard-scott, 2021-01-10:
See: https://github.com/fashberg/WThermostatBeca/issues/47

closed 2022-10-24 by github-actions[bot]