fashberg/WThermostatBeca #93: MQTT - password length problem

pery777, 2020-12-25:
Hi. I installed the latest version of hassio and started Mqtt. Mqtt generated my login and password automatically. The problem is that the password is 64 characters long and is too long to paste into the configuration mqtt wthermostat-1.18-fas
I am asking for help in solving this problem.

steptoe75, 2021-01-07:
pery777 did you find a solution to this? Im having the same trouble

closed 2022-10-23 by github-actions[bot]

labodj, 2026-11-08:
@pery777 I maintain a fork of this project and have released [v1.26.0-labo](https://github.com/labodj/WThermostatBeca-labo/releases/tag/v1.26.0-labo). MQTT passwords now support up to 64 bytes in both the web form and persistent storage. The new “Keep saved MQTT password” checkbox separates retaining a password from entering a new one: uncheck it to replace or clear the saved value. This also allows the literal password `___NOCHANGE___`.

Tests cover the 64-byte limit, punctuation, save/reload, supported legacy migration and failure handling. A 65-byte value is rejected rather than truncated. Back up long credentials before downgrading to firmware with the old 32-byte limit.