fashberg/WThermostatBeca #95: Power failure and Settings

Aryankids, 2020-12-28:
In case of a power failure, all the settings (e.g. scheduling ) go back to default.
Is this expected ? or is this a bug?

kot-gh, 2021-01-03:
I can confirm same behavior on all 5 of my BHT-002 thermostats I have, once I turn off the mains switch all of them are getting back to default scheduling temperature settings. Are these settings kept in the esp or in the mcu?

Bascht74, 2021-01-06:
Hi, can you test with 1.19.beta1 ? 
My power provider is switching off my heaters remotly and if the power went back on after 1-2 hours, the schedule is still there...

For me it looks like the devices are going to "off" sometime, if the power went back on. But I am not shure so far, as it could be the kids changing the settings in their rooms...

kot-gh, 2021-01-06:
Hi, ok, will flash them today and test

kot-gh, 2021-01-06:
Ok so I've flashed only 1 of 5 with 1.19.beta1 and it still has the same issue, all scheduler settings are set to default values after a power failure.

Bascht74, 2021-01-06:
Mmmh. 

All of my devices are not showing this behavior...

Aryankids, 2021-01-06:
same here, configured to the latest (19 beta), shut the power down to the thermostat for 10 seconds, brought it up again and the schedules are there. In other words as far as I can see the bug is fixed (Thank you). Will perform a few more test tomorrow.

Bascht74, 2021-01-17:
Hi, I tested now with a couple of devices and with one device, I can see the error from time to time.
It seams to be the MCU, because the schedule data is stored there, If I see it right.
Sometimes the device is turned off after a power restart and not turned to automatic. 

@fashberg 
Could you store the last heating mode and the actual schedule on the ESP8266 and restore it after a reboot? This way everybody can be sure that the web configured data is on the MCU and the device recovers the last mode (as with the time, as I can see it right). 
In Germany there are a couple of special power rates where the power can be turned off for some time (as you might know :-)
So this could be really helpful for some people.

Sebastian

klausahrenberg, 2021-01-23:
Hi, I'm the author of the original firmware. As far as I can see, this is a device specific issue of (hopefully only) BHT-002. I have the same issue reported at my firmware [https://github.com/klausahrenberg/WThermostatBeca/issues/148](https://github.com/klausahrenberg/WThermostatBeca/issues/148).

Until now I have no good solution. You can reconfigure the MCU at restart via MQTT broker that stored the last state. The ESP8266 itself doesnt have enough memory to store the settings twice. This is a complicated issue. My other device (ME102) does not have this behavior. That makes it more worse, because this solution would device spocific.

Bascht74, 2021-01-23:
@klausahrenberg 
Thanks for the Infos about it and the link to your issue.

Hi, I myself have those issues only with one device. 
Maybe it is firmware specific to the MCU?

I normally use Tasmota, a firmware, that is even bigger in size if you flash it (but don't work with the BHT-002). 
In there you can store three rules with hundreds of chars...
They use a compression algorithm: https://github.com/arendst/Tasmota/blob/bb76052a539ba8354a6a2b5f189448ceb2086227/tasmota/xdrv_10_rules.ino#L192
https://github.com/siara-cc/Unishox
https://www.arduino.cc/reference/en/libraries/unishox-progmem-decompressor/

The complete schedule takes - if I counted right: 135 bytes, 5 (Time) + 4 (Temp) * 3 * 6 Periods. If the Unishox can compress that with 50% (maybe more), we need maybe less than 60 bytes consistent storage...
How much free consistent memory do you have and could writing/reading the date to a free area of the flash be an alternative?
(As it is not changed often)...

btw. Thanks for your work! Great to be able to control those via HA... :-)

Sebastian

klausahrenberg, 2021-01-24:
Anyone, who is interested in a solution and like to test the provisional solution for 'deviceOn' in my firmware v1.20j - have a look here: https://github.com/klausahrenberg/WThermostatBeca/issues/148#issuecomment-766431083

closed 2022-10-21 by github-actions[bot]