fashberg/WThermostatBeca #96: [Not a Bug] BHT-002 prevent Auto Schedule possible?

itsyoshio, 2020-12-31:
Hi there!

I have a question regarding the operation of my BHT-002.
As it is now, it currently always changes to Auto Mode when a Schedule period starts (time-wise).

On #38 i've read, that the BAC has presets that can prevent this behaviour (by switching it from `scheduler` to `manual`)

Is this possible with the BHT-002 aswell, or is it not exposed to the ESP32?
Is there another way (except maybe clearing the Scheduler table) to stop the Thermostat from entering Auto-Mode?

Thanks

gavlexx, 2021-01-01:
`Is there another way (except maybe clearing the Scheduler table) to stop the Thermostat from entering Auto-Mode?`
There is option in thermostat web ui (menu Configure Thermostat) - "Switch back to Auto mode from manual at next schedule period change". It rules behavior you need to change.

itsyoshio, 2021-01-05:
Somehow I didn't see that last time. Thank you so much :)

closed 2021-01-05 by itsyoshio