fashberg/WThermostatBeca #100: Unable to flash: "Failed to connect. Timed out waiting for packet header"

chrisbaume, 2021-01-15:
I've got a BHT-002-GBLW that I'm trying to flash manually using a CH340 adapter on OS/X Catalina. I've wired up the TYWE3S chip, carefully following the instructions [here](https://github.com/fashberg/WThermostatBeca/blob/master/Flashing.md). However, `esptool.py` just gives me:

    Connecting........_____....._____....._____....._____....._____....._____....._____
    /dev/cu.usbserial-14410 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header

I've checked the physical connections using a multimeter, ensured `GPIO-0` is connected to ground, ensured `ESP-EN` is connected to `Vcc`, and tried using a separate stabilised power supply.

I used CoolTerm to see what is being received from the device, but it's a garbled mess. I've tried various different baud rates including 9600 and 115200, but I can't get anything legible from it. Using the same adapter, I can successfully use CoolTerm to communicate with an `ESP8285` with a baud of 115200.

On [this page](https://tasmota.github.io/docs/TuyaMCU-Devices/), I read about how the MCU might need to be disabled during flashing. I found the [datasheet](http://www.junshuo.net/down/html/?128.html) for my `UTF87012` and tried grounding `RST` on pin 4, with no success.

When I power the TYWE3S chip just on `ESP-VCC` and `ESP-GND`, no Wifi access point appears. Does that indicate a problem?

Many thanks for making this firmware available, and for your support!

rumoldus, 2021-01-15:
Hi ,
GPIO2 also needs to be tied to ground for flashing.
Further if no WIFI access point appears after some time (but it can take a while), you must doubt your power supply. You are using 3.3V I hope ? 5V may damage the ESP8266.
I searched for photo's, but could not find useful ones. I did take off the ESP module off the thermostat PCB, but I am not sure it is needed.

chrisbaume, 2021-01-15:
I have GPIO15 grounded but not GPIO2. Is that what you mean?

I've only used 3.3V.

Can't see any easy way to remove the module so that's be a very last resort I think.

rumoldus, 2021-01-15:
My statement may be wrong. Need to check better when I am home.
You are not keeping reset low all the time, are you ?
I had the same problem with serial output. Try something around 74000 / 78000 (Again from the top of my head).

rumoldus, 2021-01-15:
Actually GPIO2 should be pulled high ...
apologies.

chrisbaume, 2021-01-15:
Thanks, I've not seen GPIO2 mentioned before but can try that.

I've been keeping reset low. It's suggested [here](https://tasmota.github.io/docs/TuyaMCU-Devices/) that you can do that. Again, I can try removing it after boot.

Will also try those baud rates to see if they work.

rumoldus, 2021-01-15:
Suggestion is to use a resistor to pull GPIO2 high (10k). You should definitely release RST after reboot.
The baud rate of 78000 is the correct value. If I remember correctly, after flashing it changes again.
You may also need to set the baudrate in your esptool.py command line. I spent some evenings with the same problems, but eventually I got it working. I should have documented it better.

znanev, 2021-01-17:
@chrisbaume I've just dumped the original firmware of my device and flashed this one. My device is almost the same as yours - mine is BHT-002-GCLW (dry contacts relay).

The only connections that I made are as explained in that Flashing guide:

[image]

The connection to GPIO0 (blue in the picture above) doesn't even need to be permanent - you can just "touch" that pin with the free end of a connector tied to GND while applying power to the module. No need to disable the other MCU.

I haven't provided a separate power supply for the ESP module - just used the +3.3V coming from the USB-to-serial adapter.

So once you've soldered all the connections as shown above (excluding the blue connector), try the following steps:

1. Unplug your USB-to-serial adapter from your host computer
2. Touch the GPIO0 pin with a connector, tied to GND of the USB-to-serial converter at the other end
3. Plug your USB-to-serial adapter to your host computer
4. Break the connection to GPIO0 - it is not needed anymore
5. Execute the `esptool.py` command needed (adjust your port, baud rate is not important at all, you can just use the default), i.e.:

`esptool.py -p /dev/ttyUSB0 read_mac`

You should see details about your ESP module.

Following each esptool.py command execution, you should either reset the module using its RST signal, or disconnect the power supply and reconnect it back, if you need to execute another command. Don't forget to tie GPIO0 to GND briefly while resetting /repowering the module, as this is how you enter its bootloader mode.

Hope this helps!

rufik, 2021-03-05:
Unfortunetly, I'm encountering the same problem with my BHT-002 :/
It looks like I'm connected fine - I can see booting messages and MCU traffic when booted normally. I can put TYWE3S in download mode (see serial output like `ets Jan  8 2013,rst cause:2, boot mode:(1,6)`) but still getting `Failed to connect to Espressif device: Timed out waiting for packet header` error message from esptool. The MCU is keeping me out of access?

rufik, 2021-03-05:
Maybe I should disconnect pin 5 (UTX) of MCU out of the mainboard?

rufik, 2021-03-16:
I've ended up with desoldering TYWE3S module and than flashing. Otherwise no-go.
Desoldering is pretty simple, only pins VCC and EN on the left side are soldered; only pins GND, GPIO15, RX, TX on the right side are soldered.