fashberg/WThermostatBeca #99: Modbus registers?

MichielVanwelsenaere, 2021-01-15:
Hi!

Thanks for your great work and for open sourcing it.
I'm tempted to use these Beca thermostats for my setup but would like to integrate with them via Modbus RTU.
Therefor I was hoping you could shed some light on:
- do all models expose a modbus RTU interface? On every image I find I can see 'modbus' noted down on the back but I can't see where the modbus wires need to be plugged in..
- do you have a list of the exposed modbus registers and what they contain? That information seems hard to come by, even on the Beca website I can't seem to find it.

Thank you!

badger707, 2021-02-26:
It seems you need Modbus version thermostat.
There were some video on youtube where guy talks on this. Dont have link to it, sorry, but luckily I have done some screenshots at that time:
[image]

Protocol & registers
[image]

MichielVanwelsenaere, 2021-03-02:
thank you very much, closing issue.

closed 2021-03-02 by MichielVanwelsenaere