fashberg/WThermostatBeca #106: wrong more return after poweroff.

Bl00d-B0b, 2021-02-01:
Hello,

I use WThermostat 1.19.beta1-fas

when you switch to schedule (and schedule is set to lower temperature), then you power off, and then enable heating (heating is set to high temperature) mqtt return status idle, instead of heating.

Ed

xxsokolov, 2021-12-22:
+

closed 2022-10-16 by github-actions[bot]

labodj, 2026-11-08:
@Bl00d-B0b I maintain a fork of this project and have released [v1.26.0-labo](https://github.com/labodj/WThermostatBeca-labo/releases/tag/v1.26.0-labo). I reproduced the stale-action path behind the schedule → off → manual-heating sequence. State/action are now recalculated on power and mode changes even when the room temperature and target have not changed since an earlier calculation. Derived status updates also avoid sending unintended mode commands back to the MCU.