fashberg/WThermostatBeca #108: Thermostat BHT-002 sends MQTT only LWT = online

chipmoca, 2021-02-19:
Doesn't want to send temperature. I don't use a housekeeper. Help please, I have been fighting for two weeks and has not been working. We have to put another sensor next to it, on the Tasmot, it transmits the temperature normally. My experience is limited. Thanks for a great project! Sorry for my English.)

badger707, 2021-02-25:
Give us better understanding of your setup. This model is known to be working perfectly with this firmware...

chipmoca, 2021-02-26:
Thank you for responding. I am using the MQTT broker mqtt.dioty.co. One duplicate sensor on Tasmota firmware transmits temperature well. Thermostat does not want to transmit temperature, with similar settings. In Thermostat, turn the transmission function on and off in one message. In no way did it affect, I don't see the temperature. Thermostat is located in a country house, so I don't go there every day. I have already tried all firmware versions.

badger707, 2021-02-26:
You might be using incorrect topic names. For testing, you could setup MQTT broker on your local network and see if you're getting all topics readouts & you can interact with your BHT thermostat. This firmware is well documented. When you'll get it working on local network, then you can take it further to your cloud mqtt provider and seek service integration support there... Also, it is worth to check mqtt bridge setup. It all starts with firmware, then It is up to you how you want to integrate this into your ecosystem. 

On the side note, I could not make my BHT thermostat to work with my Mosquitto FQDN (local) but it works just fine with IP address.

closed 2022-10-16 by github-actions[bot]