fashberg/WThermostatBeca #109: Can't flash on BHT-002-GCLW

JinKanzaki, 2021-02-28:
So, I, for the life of me, can't get esptool to connect with the thermostat. I'm not great at soldering, so please excuse the mess but here are two pictures.
What  did I do wrong?
Does anyone know?

[image]
[image]

JinKanzaki, 2021-02-28:
I figured it out, I needed an external PSU since the FTDI adapter couldn't give me the necessary milliamps. Needs 60 mA.

closed 2021-02-28 by JinKanzaki

budisek, 2023-04-25:
Hi

may i know how did you connect TYWE3S with FTDI232? I have all wires soldered correctly. I am trying to connect it with Tasmotizer software but it can't recognize device at all.